UBC-Thunderbots/Software #16: Add custom exceptions and exception handling

MathewMacDougall, 2018-08-03:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
When we encounter run time errors, such as an unrecognized Primitive trying to be constructed, we need a better way to handle these errors than either ignoring them or immediately aborting the program. Simply using std::exit, std::abort, or assert does not give much information about the problem be default, and any case that includes once of those is not testable.

**Describe the solution you'd like**
We should add proper exception handling in the places in our code where a runtime error such as above could occur. If an unexpected scenario occurs, our code should throw an exception. Even if the exception is not handled by a parent function (which it likely won't be), it will cause the program to gracefully shut down and provide the user more context for what happened, and where. We can also include test cases that excessive these unexpected scenarios and can check the correct exception is thrown.

The first places that should be looked at are where we receive and cast Intents in the navigator (`rrt.cpp`), 
the static 'Primitive factory" function in the Abstract Primitive class (`primitive.h/.cpp`). There may be some other places marked with a `TODO` as well.

It is worth investigating the different exceptions c++ has be default. We may not need to define custom ones if there is something similar we can use with a custom message. Otherwise it should be fairly simple to define a few new custom exceptions.

**Describe alternatives you've considered**
LIke mentioned above, we can deal with runtime errors by either correcting for them as much as possible (eg. with default values), or aborting/exiting the program. We should not tolerate these errors as they are severe and indicated errors in the code elsewhere (ie. we should never be getting this unexpected data, where is it coming from?). We should also not abort/exit the program because without any print statements they provide no insight to why the program was stopped, and are not cases that can be tested with a unit test.

garethellis0, 2018-09-20:
_We're T-10 days to the end of September, so just doing a quick run-through to see if there's anything we can afford to push to an October sprint._

Is this needed for an MVP? I definitely feel it's a need-to-have in general, but I think we can afford to push this out by a month if we're not going to get it finished for the September sprint.

MathewMacDougall, 2018-09-21:
Agreed, we have higher priorities. Removing from September sprint scope.

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

This should ideally be finished for October so we can establish the "correct" procedure for the kinds of cases that require exceptions or warnings. @orayasrimokla ETA on a PR?

orayasrimokla, 2018-10-28:
Adding a pull request by the end of today! It's a very basic exception outline so it will need a little more work after October.

garethellis0, 2018-10-29:
Just making a note here that we should make sure to resolve all `TODO`'s in the codebase associated with this ticket before closing it (there are several places that we should be throwing exceptions, but are waiting for this to be finished).

MathewMacDougall, 2018-10-31:
^ Yup mentioned this in my review

PR #112 (2019-01-22): Added simple exceptions and exception handling 

### Description 
Used c++'s exception library to add exception handling. When the exception is called, it shuts down the program and provides an error message. The error message should specify where the error occurred. 

### Testing Done
Modified backend_output/main.cpp to run exceptions 

### Resolved Issues
resolves #16 

### Review Checklist
- [x] **Remove all commented out code**
- [X] **Resolve all TODO's**: All `TODO` (or similar) statements should either be completed or associated with a github issue
closed 2019-01-22 by orayasrimokla