UBC-Thunderbots/Software #22: Define how the navigator should define and query obstables

MathewMacDougall, 2018-08-05:
Our new navigator will need to handle obstacles like the old system in order to plan paths around them. We should consider how we want to handle and represent these obstacles or invalid areas.

**Option 1:**
Define a "Violation" struct like the old system, which is populated by various functions (such as Robot Violation, Ball violation, defense area...). These structs are then queried to determine if a given point is valid, or more valid than another point. The relative validity is determined by how much a point is "trespassing" into an obstacle.

**Option 2**
Since all the obstacles our navigator tends to deal with can be represented as shapes/areas, we could generalize the system to be based on areas. If we assume an area (Circle, Rectangle, etc.) implements a function like `getDistanceInside` or `pointInside` then we could perform the same operation on all obstacles.

Example:
* provide the navigator a list of areas (obstacles). These obstacles may be inflated to account for the extra "avoid distance buffers" specified by our parameters. For example, we may want to avoid Robots by 10cm, meaning we need to add 10cm to the radius of a normal robot when we create the area.
* Determine the Violation of a given point by calling `getDistanceInside(Point)` on all the area obstacles
* Then determine if a Point is valid or not based on the Violation amount

Other options should also be considered.

henrymliu, 2018-08-18:
I am in favour of option 2, with some minor alterations:
We have two classes of obstacles; robots and field boundaries.
Robots are represented as circles, and an additional Segment/Vector that points in the direction of travel, with length proportional to velocity.

Field boundaries are represented as rectangles.

MathewMacDougall, 2018-08-19:
Sounds good to me :+1:

PR #39 (2018-09-04): Add initial obstacle interface
Fixes #22 

~~Still a WIP as there is a weird link error with the dist() function used in obstacle.cpp, even though it is defined and implemented properly...~~

Added a new RobotObstacle class to handle robots (with capability for collision avoidance in the future), and added two new member functions to the Field class so the defense areas can be easily retrieved as Rects. I will also add the field boundary later.

For now, we will use an avoid distance on the RobotObstacles to determine how far to avoid other robots. I thought for a really long time about bringi [... cut]
closed 2018-09-04 by henrymliu