UBC-Thunderbots/Software #43: Create structure to read RefBox data

MathewMacDougall, 2018-09-08:
In addition to SSL vision data that gives us the state of the world (robot positions, etc.) we also need to listen for commands from the Referee Box.

We should create a simple datastructure to store the refbox data (so we can abstract away how we get the information), and this structure should be set up to be passed down to components that will use the information.

The refbox from previous years can be found here: https://github.com/RoboCup-SSL/ssl-refbox
There had been talk about re-writing it for RoboCup 2019. The new repository can be found here: https://github.com/RoboCup-SSL/ssl-game-controller

The repository contains some reference clients for receiving messages, so that would be a good place to start. Likely the structure can be similar to how we receive ssl-vision messages in the `backend_input` node

garethellis0, 2018-09-20:
_We're T-10 days to the end of September, so just doing a quick run-through to see if there's anything we can afford to push to an October sprint._

Obviously we can't play without this, but maybe we could take it out of scope for the MVP if it's not reasonable to get it done in next 10 days?

MathewMacDougall, 2018-09-21:
Agreed, not essential for September MVP. Bumping for now, and if someone has extra bandwidth it can be started early.

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

Ideally we should have this done for October, since it's a pretty direct feature of this Milestone's goal of a connected system. @jordanzliu seemed like you were close on this. ETA on a PR?

jordanzliu, 2018-10-25:
PR for this one should be up by end of Saturday, *should* have enough time to put up a PR for #72 by the 31st