UBC-Thunderbots/Software #52: Make predictors only take positive time values

MathewMacDougall, 2018-09-14:
Based on what I've previously seen in our code usage, and some further thought, I don't think we (currently) have a use for accepting negative time values in functions with prediction capability.

For example the `Ball.position(double time_delta)` function, and similar functions for `Robot`.

These functions should only accept positive values because
* We don't have a use for getting past values at this time
* We should not be applying the same "prediction" algorithms backwards in time, since we already know where a robot/ball was based on the input data from vision

This seems reasonable to do as part of #49

garethellis0, 2018-09-20:
_We're T-10 days to the end of September, so just doing a quick run-through to see if there's anything we can afford to push to an October sprint._

I don't think this is needed for MVP? Unless it's already partially done, I'd say push it out.

MathewMacDougall, 2018-09-21:
Not needed, but it's already getting done as part of the timestamp updates.

MathewMacDougall, 2018-10-03:
Was resolved in #56 and #54

closed 2018-10-03 by MathewMacDougall