UBC-Thunderbots/Software #63: Implement MRF Backend

henrymliu, 2018-09-22:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
We kind of need an MRF backend to communicate with the physical robots.

**Describe the solution you'd like**
Port the MRF backend from the old code.

**Describe alternatives you've considered**
We could also rewrite it from scratch, but that would likely take a long time. 

**Additional context**
Breaking this down a bit:
- There is a libusb wrapper in `util/` that we should consider porting too. Additional classes in `util/` such as `NonCopyable` may also be useful to port over. Need to figure out where to put these.
- `Annunciator` is used for conveying messages from the field (ex. dead robots, hardware issues, basically whatever makes the dongle beep); this may have to be ported too and integrated into whatever logging system we come up with
- The old backend directly communicates with a `MRFRobot` instance; since we no longer have this and instead are using primitives only, this will have to be changed.

henrymliu, 2018-09-22:
Will update this more as I discover new things.

MathewMacDougall, 2018-10-25:
Bumping to November milestone as this isn't critical for October. @henrymliu how's progress?

henrymliu, 2018-10-26:
I think I can theoretically attach to the dongle now (need to verify on Saturday). Close to having something that can send primitives to the robots.

henrymliu, 2018-11-18:
@MathewMacDougall good news: fixed the unfinished transfer issue, now we are officially able to send drive packets

bad news: this involved a hack, namely turning the backend_output node into a GTK application because the libusb wrapper callbacks use glib, which basically needs the GTK main loop :(

I think the long term plan would be to make a separate thread that handles libusb callbacks. This probably also means some synchronization, but it shouldn't be too bad.
In addition, we still need to figure out a) the vision thread and b) how the dongle is going to be shared across backend_input and backend_output

MathewMacDougall, 2018-11-19:
That's great! You are a god.

Unfortunate about the hack, but at least it's isolated to a single, relatively small node (yay new architecture). I didn't realize `glib` was so reliant on `GTK`.

The vision thread is easy, assuming we can send vision packets in addition to drive packets (I remember they were different). The `backend_output` node can simply subscribe directly to the ball/team data published by the `backend_input` node and create a packet from that. That is the critical path between data input/output that skips all the AI logic processing time.

For dongle sharing, especially considering the potential complexity of these callbacks and synchronization, it might be worth putting the dongle entirely in it's own node. The dongle node can then listen for `Primitive` messages (or something even simpler, maybe a pre-encoded packet), and publish messages it receives from the robots. Can we receive messages yet?

@garethellis0 for thoughts on separating the dongle into it's own node.

garethellis0, 2018-11-19:
Hmm, separating the dongle into it's own node seems like a fairly reasonable option, but I'd like to see what the hacky implementation looks like first. @henrymliu - could you point me at the branch you're doing this on (presumably on your fork)? 

With regards to the glib/gtk coupling, at least theoretically, I don't think they're really linked at all? Gtk might make it _easier_ to use glib, but we should totally be able to use it without it (I would think).

henrymliu, 2018-11-19:
@MathewMacDougall I figured that was the only major change we need before supporting vision packets, I don't think that is too difficult to implement (although i might be busy for the rest of the term). 
Messages are... technically being received I think? It's just that I have nowhere to send them to for now (stuff like charger state, estop state, robot statuses). 
I'll have to think about the dongle node option a bit more, although it seems to work.

@garethellis0 Branch is on https://github.com/henrymliu/Software/tree/mrf_backend_test
The hack is in main.cpp in the backend_output node; basically we init a GTK application and start it in a separate thread before the ROS loop.

garethellis0, 2018-11-20:
Oh man, there's a LOT of stuff here. I'm really struggling to get a handle on what's going on just because of the sheer volume of the diff. Not sure how you wanted to do this, but I'd suggest chopping that branch down to the minimum required stuff, then hopefully it should be a bit easier to figure out how we can do this without gtk.

henrymliu, 2018-11-20:
Yep, I plan to cut some more stuff out later (esp. `Annunciator`, maybe `Property` too). The dongle class definitely could be trimmed/split up as well.
The `libusb` wrapper is also heavily integrated with GLib, so that's the first point to start decoupling.

PR #316 (2019-02-23): (3/4) Add MRF Dongle class

## Please fill out the following before requesting review on this PR

### Description

Currently this is blocked by #333; the additional changes here will disappear once that is merged.
Adds the dongle class that initializes communication with the radio dongle.
### Testing Done

Field testing! Again unit/integration testing this may be difficult and restricted to the packet encoding functions.

### Resolved Issues

Partially resolves #63

### Length Justification

This is a large and complex class that handles both input and output communication, splitting this up f [... cut]
PR #335 (2019-02-24): (4/4) Implement MRF Backend

## Please fill out the following before requesting review on this PR

### Description
Adds a backend that sends primitives over radio.
The diff is large because it contains changes from #333 and #316, this should reduce after those PRs are merged.

### Testing Done
Field testing! Relies on legacy functions for encoding, which are not easily testable.

### Resolved Issues

Resolves #63 
### Length Justification

### Review Checklist
closed 2019-02-03 by qgolsteyn