UBC-Thunderbots/Software #74: AI GUI MVP - Visualize the field

MathewMacDougall, 2018-10-03:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
Once we decide on how we will be implementing a GUI for our AI (see #73 ), the first step / MVP would be to visualize the field. This would just be the field from the image below, without the other controls or logs. The robots and ball do not need to be drawn on the field.

[image]

**Describe the solution you'd like**
Draw the Field in the GUI using the ROS messages published by the `backend_input` node and the framework we decide on from #73

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

Ideally we should have this done for October, since it's a pretty direct feature of this Milestone's goal of a connected system. Or at the very least, the real "core" of the visualizer. @qgolsteyn still on track for a PR by Oct 27th?

qgolsteyn, 2018-10-25:
Yes, it is still on track for October 27th.