UBC-Thunderbots/Software #75: AI GUI - Visualize the ball on the field

MathewMacDougall, 2018-10-03:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
Once #74 is complete, the next step is to visualize the friendly and enemy robots on the field, using the friendly and enemy team ROS messages published by the `backend_input` node. See the image below for an example of our old AI.

[image]

**Describe the solution you'd like**
Draw the friendly and enemy robots in the GUI using the ROS messages published by the `backend_input` node. Depends on #74

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

Ideally we should have this done for October, since it's a pretty direct feature of this Milestone's goal of a connected system. Or at the very least, the real "core" of the visualizer. @qgolsteyn still on track for a PR by Oct 27th?

qgolsteyn, 2018-10-25:
Yes, it is still on track for October 27th.

closed 2019-01-10 by qgolsteyn