UBC-Thunderbots/Software #81: Revise ros-latest.list in Software Setup

jessicahe21, 2018-10-06:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
1. When running `./setup_software.sh melodic` on Linux Mint 19, the error below appears and is then followed by failure to install ROS.
```
E: The repository 'http://packages.ros.org/ros/ubuntu tara Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
2. After resolving the first issue, running the script again returns an error indicating that installing ROS dependencies has failed and that Mint is an unsupported OS.

**Describe the solution you'd like**
1. On Line 134 (kinetic) and Line 147 (melodic) of `setup_software.sh`, replace `$(lsb_release -sc)` with `bionic`. 
Previously, `apt-get update` fails because `ros-latest.list` is set incorrectly by the script.
2. First run `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list`, then add `--os=ubuntu:bionic` at the end of Line 176.

*(Thanks to Henry for the help and solutions!)*

**Describe alternatives you've considered**
If this is a Linux Mint only issue, a separate `setup_software.sh` with the above changes for Mint users might be cool.

**Additional context**
This is my first time posting on GitHub, I decided against making a pull request as I wasn't 100% confident. Please let me know if this is inappropriate (i.e. not important enough to create an Issue for miscellany things like this).

henrymliu, 2018-10-06:
So the distro (xenial or bionic) can be parameterized, e.g. we can have a variable `distro` that can be set to either `xenial` or `bionic` depending on the ROS version chosen, and then we can use this in `--os=ubuntu:$distro` and replace `$(lsb_release -sc)` with `$distro` for example.
`sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` is not needed, I just wanted you to run that for `rosdep init` to be run again.

henrymliu, 2018-10-06:
I've also updated the title generalize the issue a bit more.

jessicahe21, 2018-10-07:
Thanks for the info and update, that made a lot of sense! Edited the original comment.