UBC-Thunderbots/Software #87: Create Shoot Primitive class

MathewMacDougall, 2018-10-13:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
We need a class in the AI for the Shoot Primitive, so the AI can use it as an output.

**Describe the solution you'd like**
Implement a Shoot Primitive class that extends from the base Primitive class. This will go in the `thunderbots/software/ai/primtiive` folder. See the `move_rimitive.*` files for an example.

Make sure to document the spec of the primitive, using the information from #86. This can probably go at the top of the class (in a big javadoc style comment).

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

How's progress on this? It should be a bit easier now that #86 is done. Let me know if I can clarify anything.

jessicahe21, 2018-10-27:
Apologies on the delayed response. Goal is to have it down by this weekend and test it in the following week.

MathewMacDougall, 2018-11-13:
@jessicahe21 it would be great if you could also add some unit tests for this class as well. That will give us test coverage right away, and is a good introduction to testing in our codebase.

You can use #131 as an example. Sorry for the last minute change.

PR #122 (2018-12-12): Added Kick and Chip Primitive classes

## Please fill out the following before requesting review on this PR

### Description
Added a ~~Shoot~~ Kick and Chip Primitive classes under ~~`shoot_primitive.*`~~ `kick` and `chip_primitive.*` that extends from the base Primitive class. 

Follows similar formatting as `move_primitive.*` such as the commenting style. 
(This is a follow-up PR to #121 )

### Testing Done
Unit tests are done for all KickPrimitive and ChipPrimitive class functions.
Running `catkin_make` builds the code / AI.

### Resolved Issues
resolves #87 , part of #135 

### Review Chec [... cut]
closed 2018-12-12 by jessicahe21