UBC-Thunderbots/Software #91: Create DirectVelocity Primitive class

MathewMacDougall, 2018-10-13:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
We need a class in the AI for the DirectVelocity Primitive, so the AI can use it as an output.

**Describe the solution you'd like**
Implement a DirectVelocity Primitive class that extends from the base Primitive class. This will go in the `thunderbots/software/ai/primtiive` folder. See the `move_primitive.*` files for an example.

Make sure to document the spec of the primitive, using the information from #86. This can probably go at the top of the class (in a big javadoc style comment).

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

How's progress on this? It should be a bit easier now that #86 is done. Let me know if I can clarify anything.

HaomiaoZ, 2018-10-28:
Hey Mathew, I do have question about the Direct Velocity Class. For direct velocity class, all we need to do is to create constructors from either thunderbots_msg or some given values and implement all the virtual functions in primitive.h?

MathewMacDougall, 2018-11-01:
Hey sorry for the late reply. What you said is correct.

MathewMacDougall, 2018-11-13:
@HaomiaoZ it would be great if you could also add some unit tests for this class as well. That will give us test coverage right away, and is a good introduction to testing in our codebase.

You can use #131 as an example. Sorry for the last minute change.

PR #120 (2018-11-25): Direct velocity primitive class
### Description
Create Direct Velocity Primitive class as extension of Primitive class. Similar to move primitive, two types of constructor are created and all the functions in the primitive class are overridden. 

### Testing Done
Unit test can be complied through `catkin_make` , and all unit tests pass.

### Resolved Issues
resolves #91 

### Review Checklist
***It is the reviewers responsibility to also make sure every item here has been covered***
- [x] **Start of document comments**: each `.cpp` and `.h` file should have a comment at the start of it. See fi [... cut]
closed 2018-11-25 by HaomiaoZ