UBC-Thunderbots/Software #92: Create DirectWheels Primitive class

MathewMacDougall, 2018-10-13:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
We need a class in the AI for the DirectWheels Primitive, so the AI can use it as an output.

**Describe the solution you'd like**
Implement a DirectWheels Primitive class that extends from the base Primitive class. This will go in the `thunderbots/software/ai/primtiive` folder. See the `move_primitive.*` files for an example.

Make sure to document the spec of the primitive, using the information from #86. This can probably go at the top of the class (in a big javadoc style comment).

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

How's progress on this? It should be a bit easier now that #86 is done. Let me know if I can clarify anything.

MathewMacDougall, 2018-11-13:
@SupoWupo it would be great if you could also add some unit tests for this class as well. That will give us test coverage right away, and is a good introduction to testing in our codebase.

You can use #131 as an example. Sorry for the last minute change.

MathewMacDougall, 2018-12-03:
Bumped this to the December milestone. I'd like to get this done before Christmas. @SupoWupo any updates?

SupoWupo, 2018-12-04:
I've created all the primitive classes and I'm working on the unit tests. My finals schedule is kind of unfortunate since I have finals until the 19th. I might be able to work on it but I don't know if I'll have enough time so I'd appreciate some help 👍

PR #151 (2019-01-26): Roark/added directwheels and movespin primitives

## Please fill out the following before requesting review on this PR

### Description
*added the movespin and directwheels primitives*

### Testing Done
*completed unit tests for the added primitives*

### Resolved Issues
*resolves #89, resolves #92*

### Review Checklist
*(Please check every item to indicate your code complies with it (by changing `[ ]`->`[x]`). This will hopefully save both you and the reviewer(s) a lot of time!)*
***It is the reviewers responsibility to also make sure every item here has been covered***
- [x] **Start of docu [... cut]
closed 2019-01-26 by SupoWupo