UBC-Thunderbots/Software #94: grSim backend - Create Pivot Primitive

MathewMacDougall, 2018-10-13:
[grSim](https://github.com/RoboCup-SSL/grSim) is the "standard" simulator for the SSL Division, and is the simulator we use.

Because we currently don't/can't simulate our robot's firmware, we need to re-create how we move and control the robots for the simulator. Rather than sending a Primitive command to a robot over radio, which will then be executed by the robot firmware in order to do whatever the command asks, we need to send different commands to the simulator to move the robots.

We already have the ability to send commands to the simulator with #12 . The ability to control a robot's movement is further abstracted by the motion controller developed in #9 . This imitates the bang-bang motion controller on our robots, so that robot's will speed up and slow down more realistically as the move in the simulator.

**Currently blocked by #9 **

The task here is to create a function/routine to implement our Pivot Primitive using this motion-controller abstraction, so this Primitive can be simulated. We need these primitive to work in the simulator so we can actually use the simulator for testing.

The spec for the Pivot Primitive can be found in #86 , and also in the PivotPrimitive class from #88 

The PivotPrimitive class will be the input. The exact place this code will go will be revised once #9 is merged, and we will update the ticket.

garethellis0, 2019-03-02:
@bryanlogue - can you get this done this week?

MathewMacDougall, 2019-03-12:
@bryanlogue any updates here? This ticket has been inactive for quite a while. I'm going to reassign it if you're not able to make progress / finish it really soon, as it really needs to be done now.

MathewMacDougall, 2019-03-18:
Re-assigning this as no progress is being made.

PR #527 (2019-04-06): grSim Pivot implementation

## Please fill out the following before requesting review on this PR

### Description

A sort of working implementation of pivot given the state of the current motion controller. 
### Testing Done

Tested in grsim, works well enough for what we need it to do, pivots and stops.  

### Resolved Issues

Resolves #94 

### Length Justification

### Review Checklist
closed 2019-04-06 by akhilveeraghanta