UBC-Thunderbots/Software #104: Create Dribble Primitive class

MathewMacDougall, 2018-10-18:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
We need a class in the AI for the Dribble Primitive, so the AI can use it as an output.

**Describe the solution you'd like**
Implement a Dribble Primitive class that extends from the base Primitive class. This will go in the `thunderbots/software/ai/primtiive` folder. See the `move_primitive.*` files for an example.

Make sure to document the spec of the primitive, using the information from #86. This can probably go at the top of the class (in a big javadoc style comment).

MathewMacDougall, 2018-10-25:
_We're T-7 days to the end of October, so just doing a quick run-through to see if there's anything we can afford to push to an November sprint._

How's progress on this? It should be a bit easier now that #86 is done. Let me know if I can clarify anything.

MathewMacDougall, 2018-11-13:
@MaxonZhao it would be great if you could also add some unit tests for this class as well. That will give us test coverage right away, and is a good introduction to testing in our codebase.

You can use #131 as an example. Sorry for the last minute change.

MaxonZhao, 2018-11-17:
@MathewMacDougall Sorry for the late response. I couldn't make it to the previous two meetings but I'm working on it and almost finished. I will try to add unit test today.

MathewMacDougall, 2019-01-05:
@MaxonZhao any updates?

MathewMacDougall, 2019-01-11:
@MaxonZhao how's progress? Anything I can help with or clarify?

PR #261 (2019-02-18): Dribble primitive

## Please fill out the following before requesting review on this PR

### Description
Moves the robot in a straight line from its origin to destination with dribbler on. When robot arrives at destination, it will arrive with a given orientation and a specified final speed. The rpm of the dribbler as well as a boolean that describes if a small kick is allowed will also be provided.

Parameters (in order):
1. Destination x and y coordinates. Point object, units in meters
2. Destination orientation (in global coordinates), units are radians
3. Final speed units in m/s
4. Dribbl [... cut]
closed 2019-02-18 by MathewMacDougall