UBC-Thunderbots/Software #125: Create Catch Primitive class

matthewberends, 2018-11-03:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
We need a class in the AI for the Catch Primitive, so the AI can use it as an output.

**Describe the solution you'd like**
Implement a Catch Primitive class that extends from the base Primitive class. This will go in the `thunderbots/software/ai/primtiive` folder. See the `move_primitive.*` files for an example.

Make sure to document the spec of the primitive, using the information from #86. This can probably go at the top of the class (in a big javadoc style comment).

MathewMacDougall, 2018-11-13:
@matthewberends it would be great if you could also add some unit tests for this class as well. That will give us test coverage right away, and is a good introduction to testing in our codebase.

You can use #131 as an example. Sorry for the last minute change.

matthewberends, 2018-11-17:
Yes I can do that. The tests should be ready for review is in the PR for this issue.

closed 2018-11-24 by matthewberends