UBC-Thunderbots/Software #143: Cleanup and migrate all files for motion

Tim1520, 2018-11-24:
**Files Include**
motor.c/h
wheels.c/h
control.c/h
physics.c/h
dribbler.c/h

**Needs to be completed**
Add clear descriptions function descriptions where absent
Update parameter and function names to reflect their purpose
Resolve or better define "TODO" comments in functions
Standardise format

@garethellis0

garethellis0, 2018-11-24:
This issues looks pretty good! The only comments I have would be:
- Might be good to add a quick description to the top of the issue saying something like: "Migrate over motion related firmware files to this repository"
- do you mean "formatting" when you say "format"?
- For lists, github will let you format them with nice bullet points, see here (https://guides.github.com/features/mastering-markdown/) for details
- I'm created a github "Project" so that we can link all the issues related to the firmware migration together (you can see what "Project" this issue is in on the right-hand side)

MathewMacDougall, 2018-11-24:
@garethellis0 should we make this a project or milestone? I believe I turned off the "projects" tab in the main repository since I figured it was best to just use 1 simple system.

Also @Tim1520 should I move the old firmware to the `Software` repo then?

garethellis0, 2018-11-24:
@MathewMacDougall that's a great point, can we have issues in multiple milestones? If so then I definitely agree.

MathewMacDougall, 2018-11-24:
@garethellis0 unfortunately looks like issues can only be part of a single milestone

MathewMacDougall, 2018-11-24:
In what case do you see issues needing to be part of multiple milestones? For the firmware cleanup, I would see that as a separate milestone, and those tickets would be labeled `firmware` as well to distinguish them.

garethellis0, 2018-11-24:
@MathewMacDougall - the idea here was to move over the firmware from the old repo to the new one part-by-part, cleaning it up as it we go. 

Hmm, my only thought is that these tickets are all part of the "firmware_migration", so was looking for something to link them up.

garethellis0, 2018-11-24:
Sorry, the seperate milestones thing was to keep our nice monthly milestones, with another milestone to track the overall progress of the firmware migration. I don't feel super strongly about this, so I'd be fine just tagging them as "firmware" and removing the project I made.

garethellis0, 2018-11-24:
Let's put it under a new tag "firmware_cleanup" to keep them grouped.

closed 2019-07-24 by MathewMacDougall