UBC-Thunderbots/Software #153: Extremely high CPU usage

qgolsteyn, 2018-12-30:
**Describe the bug**
On my Ubuntu machine (Surface 4 Pro with Ubuntu 18.04), running the Thunderbots nodes (ai_launch, backend_input, backend_output) uses 79% of my CPU. This is without running grSim.

**To Reproduce**
Steps to reproduce the behavior:
1. In a console, run `roslaunch thunderbots ai.launch`

**Expected behavior**
Lower CPU usage from these three nodes.

**Screenshots**
[image]

**Additional context**
My machine specs:
Surface Pro 4
CPU: Intel Core i5-6300U (2.4 to 3.0 GHz)
GPU: HD 520
RAM: 4GB
Swap: 8GB
Storage: 128GB

MathewMacDougall, 2018-12-30:
This is a known issue. This happens because right now the nodes don't have that much work to do so are looping at full speed. Also because the `backend_input` node is polling the network for vision packets, it's pretty much like running a `while True` loop most of the time.

Another reason this uses an unexpectedly high percentage of the CPU is because this is 3 separate executable (1 for each node) all running full speed at the same time, it's not just a single thread in a single executable.

As more logic gets added I expect the nodes to slow down, and we can cap their looping rate as we feel necessary.

I think it's more appropriate to create tickets for:
* Receiving vision packets based on a callback/interrupt from the network. We should plan how this will interact with the potentially new radio-dongle node. We should probably have a meeting about this in the New Year since that should be the last major architecture change (at least ROS-wise).
* Setting maximum loop rates on each node if we think it's necessary (even if it's just to prevent essentially running `while True`). This may affect timing constraints, so might be a good secondary discussion to the above point

@garethellis0 @henrymliu @akhilveeraghanta for visibility

garethellis0, 2018-12-30:
I agree, and it's worth noting that the way the scheduling works out it doesn't usually end up slowing down other processes too much (in my experience).

qgolsteyn, 2018-12-30:
It's not a huge issue in the grand scheme of things, I agree. On my Surface, it does end up slowing my machine somewhat, especially when I have grSim + VS Code open when developing the visualizer. Again, not a huge deal, but would be good to get fixed at some point.

(If anything, it helps my Surface reach a nice balmy 90 degrees Celsius, which is always good to stress test my fan)

MathewMacDougall, 2019-01-06:
The final decision regarding this issue was that since we want the nodes to be processing information as fast as possible, we will not rate-limit the nodes.

The `backend_input` node _does_ receive network data asynchronously, but the data that is cached is still polled continuously which causes the heavy CPU usage with this node. We should be able to re-write this node to only publish info when the network receives a packet (without the polling). I'm made #155 to track this issue.

closed 2019-01-06 by MathewMacDougall