UBC-Thunderbots/Software #154: Add Flags to Move Intent

jonl112, 2019-01-05:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
Old AI had special move actions, such as move_dribble. This feature needs to be added to new AI. 

**Describe the solution you'd like**
Add these features as flags to move intents, intending them to be consumed by the navigator or be passed to the primitive level.

**Describe alternatives you've considered**
We could make specific actions to describe the function of each flag, but unifying the actions into one move intent with flags allows more code reuse.

henrymliu, 2019-05-04:
After discussions, this is not needed.

henrymliu, 2019-05-04:
Would it be useful to add this to the Intent base class, not just MoveIntent?

MathewMacDougall, 2019-05-06:
Hmmm not sure. Traditionally, the flags were only really for the MoveIntent since it was the only one that incorporated navigation. ATM I can't think of a use for the flags in other intents, so for now it's probably easiest to just add them to the MoveIntent. It should be easy to make generic in the future if we need it.

henrymliu, 2019-05-07:
Currently I'm porting over MoveFlags from the old AI; if we ever want other flags for the other intents in the future I should consider that in the design.

henrymliu, 2019-05-07:
Maybe if it's so trivial to port we might as well do it in the first place?

MathewMacDougall, 2019-05-17:
Fine with me

PR #597 (2019-05-23): MoveIntent MoveFlags

## Please fill out the following before requesting review on this PR

### Description

Ported over MoveFlags from old AI, and added them to MoveIntent. If I did anything stupid...  blame jetlag

### Testing Done

unit tests

### Resolved Issues

resolves #154 

### Length Justification

### Review Checklist
closed 2019-05-23 by henrymliu