UBC-Thunderbots/Software #161: Assign friendly robots based on friendly team color

MathewMacDougall, 2019-01-06:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
Logic should be added to change which robots are on the friendly team based on the `friendly_team_color` set in the Parameter Server (see #160)

**Describe the solution you'd like**
Implement logic to assign robots to the friendly team based on their color and the color of the friendly team. This should go in the `backend_input` node.

henrymliu, 2019-05-18:
@akhilveeraghanta i'm going off hazy memory of a discussion at the beginning of this month, is this okay to move to the backlog or are the requirements clear now?

akhilveeraghanta, 2019-05-20:
@henrymliu we need to check if the refbox provides this information, which would change the scope of this ticket

henrymliu, 2019-05-20:
Briefly looking at `ssl_referee.proto`, there are `TeamInfo` messages for yellow and blue teams, and `TeamInfo` contains a `name` field for the name of our team. Those can be used to determine what color we are.

closed 2019-07-24 by MathewMacDougall