UBC-Thunderbots/Software #165: Add logic to start/stop the AI

MathewMacDougall, 2019-01-06:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
The AI needs logic to start/stop running it's decision making.

**Describe the solution you'd like**
Logic should be added to the `ai` node to start/stop running the AI logic based on the `AI_running_state` from the parameter server. See #164.

akhilveeraghanta, 2019-04-19:
@MathewMacDougall now that the AI doesn't run in a loop and is more callback based, do we need this param + check? 

If so, would we just check at the begining of `worldUpdateCallback` and return if `run_ai` is set to false?

MathewMacDougall, 2019-04-19:
Good point. Yes that seems more reasonable.

closed 2019-04-28 by akhilveeraghanta