UBC-Thunderbots/Software #166: Add AI control parameters for the High-Level module in use

MathewMacDougall, 2019-01-06:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
AI needs a way to allow external modules (like the Visualizer) to control which High-Level module is in use.

**Describe the solution you'd like**
* Create a structure/constants for the possible HL modules that can be used, whose values are strings (Eg. "STP", or "STPChooseable")
* Create a function that returns a list of all the available HL modules, as strings
* Create a parameter in the Parameter Server (depends on #82) for the HL modue that should be in use

**Describe alternatives you've considered**
One alternative was to allow the HL modules to be set with a ROS Service, but the Parameter Server was deemed more appropriate and generic.

garethellis0, 2019-02-22:
Is this required for the "Competition Time!" release?

MathewMacDougall, 2019-02-22:
No, since we will only have one "High Level" implementation by then (STP)

MathewMacDougall, 2019-03-14:
Closing as this is not a relevant ticket to the work in our short/medium term future (2 years).

closed 2019-03-14 by MathewMacDougall