UBC-Thunderbots/Software #170: Add global Logger-like Singleton to allow for generic "draw shape" calls from the AI

MathewMacDougall, 2019-01-06:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
The AI needs a way to draw generic shapes on the "field" in the Visualizer, to visualize various useful data (such as ideal locations to pass, highlighting the most threatening enemy, etc.)

**Describe the solution you'd like**
Create a logger-like Singleton that has functions to draw various shapes in the Visualizer.
* The singleton would have a categories of things that can be drawn. Eg. "Pass data", "Navigator Path Data", "Enemy threat". This would allow the visualizer to easily toggle certain categories being shown or not (rather than turning off _all_ lines being drawn).
* The functions should all accept a geom object (like `Seg`, `Circle`, `Point`), colour specified by an RGBA value (4 `uint_8` values), and the category the drawing belongs to. Eg `Drawer::DrawCircle(Circle(Point(0, 1), 4.5), 255, 0, 0, 0, ENEMY_THREAT);`
* The Singleton should publish to a separate topic for each category
* The Singleton should aggregate all the Draw calls into arrays and only publish a single message per loop/tick in its respective node
* Even if no Draw calls were made, the Singleton should still publish an empty array of shapes
* Message types for the shapes and colors will be required. ROS should have default shape messages.

Note: It may be necessary to add a ROS service that returns a list of all the available categories (as strings) to the Visualizer can be aware of all the categories (so it can prepare GUI elements to enable/disable each of them)

qgolsteyn, 2019-01-08:
Current thoughts:

I think in this case a regular class would make more sense, with one instance of a `DrawLayer` (what we should perhaps call it) per layer (what you are calling a category).

Would prefer not letting colour being specified by an rgba value. Would rather see a colour constant being chosen (red, orange, green, etc.) to allow the visualizer to make the final call on what that colour should be.

Any way we could limit the number of shape messages to 60 per seconds? Anything more is a waste of resources.

Available layers (what you call categories) can be inferred from the available topics. If a topic has a valid message type, it's a valid layer. The visualizer can detect and subscribe to it. No need for a separate service.

In terms of how it could look in the visualizer, here is a mockup of what I was thinking so far:
[image]

MathewMacDougall, 2019-01-08:
I like the mockup.

I agree with your "one topic per object that needs to be drawn" proposal. There would be a unique message type/topic for each object, but the interface on the AI side would likely be member function of a singleton (so that it can be accessed from anywhere).

The reason I was thinking of specifying colors as `rgba` values is because that information can directly be used to create colors. Using color enums means that's another datatype we need to keep "in sync" between the AI and Visualizer. Not using enums saves us having to check if a value exists/is supported by the visualizer. Not that I expect colors to be changed/modified much (at all), that was my reasoning.

As far as I know there isn't an easy way to throttle messages on the `AI` side because publications will be in-sync with the loop rate of the AI nodes, which we want to run as fast as possible. It's possible but IMO not as nice/deterministic to do some kind of time comparison to only publish after 1/60 of a second has passed. It already seems like the Visualizer has the capability to drop/throttle messages above 60HZ, so I think the main concern here would be network bandwidth? Or is there another reason the AI should handle the throttling vs the Visualizer?

MathewMacDougall, 2019-01-08:
pinging @garethellis0 and @akhilveeraghanta for ROS wisdom and second opinions.

garethellis0, 2019-01-08:
- Ideally we try to get away without any drawing class at all. I think the ideal here would just be a function like `drawShapeOnTopic(Shape s, Topic t)` that we could call. The only issue with this might be the delay in setting up the publisher (I'm not sure how significant it would be). The main point here is allowing the user the flexibility to publish a shape to any topic at any point without prior setup. If the publisher creation does turn out to be slow, then we could always to a map of `topic_name`->`publisher` in our drawing singleton class
- I support rgba values, I think it reduces complexity overall (principally for the reasons Mathew mentioned above)
- I don't think we should worry about throttling messages until it becomes an actual performance concern, as it could add significant complexity for minimal gain (though it would be good to explictly measure the performance of publishing many shapes to make sure it's not a concern) Network bandwidth isn't really a concern since we're talking over some internal bus in the computer.
- I like the mockup! :+1:

MathewMacDougall, 2019-01-08:
@garethellis0 I like your idea of a generic `drawShapeOnTopic` function (I realized I was originally assuming that there would only be one shape used on each topic).

However we will still need a singleton in this case, so that we can aggregate the shapes/drawings and send them all at once at the end of the `main` loop in the node. Otherwise the shapes would overwrite the previous one if the Visualizer draws once per publish, and removes the previous drawings.

garethellis0, 2019-01-09:
Ah, yeah, that makes sense, especially if we want to keep code ROS agnostic.

akhilveeraghanta, 2019-01-09:
Is there any benefit to using a new topic for any new shape we would like to draw? Can we just send everything on one topic so that the visualizer doesn't need to parse through the list of topics and find the ones that it should listen to every tick? 

We could just have a custom `draw` msg with separate arrays the singleton will add the data to, to distinguish between the layers/filters, and publish that msg on one topic?

garethellis0, 2019-01-09:
The thought here was that one graphical thing per topic was that it gives a nice abstraction between the visualizer and the AI. The visualizer won't have to worry about draw order for objects on a given topic, and we can setup the visualizer so that we can make explicit choices about what layers/topics show up in top of other layers/topics.

MathewMacDougall, 2019-01-09:
I think the main debate here is how color is going to be specified

**Option 1:** Pass color values as `rgb` values
**Option 2:** Use an enum/constant for color values (eg. `RED`, `GREEN`, `CYAN` etc.)

**Why use Option 1 over Option 2**
* No extra datastructure is required that has to be kept (in sync) between the AI and Visualizer
* `rgb` is a "well defined" thing so should be easy for the Visualizer to render colors generically
* AI has easy access to the full color spectrum

**Why use Option 2 over Option 1**
* It's easier to look at a `Draw` call in AI and know what the color is. `RED` is more clear than `(230, 14, 45)`, especially for weird colors
* The Visualizer has control over how the colors are exactly rendered. Useful to make the colors match the rest of the system, and for things like a "dark mode"
* Related to the above, allowing the Visualizer to determine exactly how colors are displayed is a better separation of concerns

Here's my summary of the arguments from both sides. @qgolsteyn @garethellis0 please correct me if a missed anything.

I'm personally leaning towards option 1 because I feel it's simpler and more generic. Although the arguments for Option 2 are valid, I'm more strongly prioritizing simplicity over aesthetics.

@qgolsteyn @garethellis0 @akhilveeraghanta @henrymliu @EvanMorcom for your opinions

garethellis0, 2019-01-10:
I'm still for option 1. I'm sure we can come up with a way to nicely abstract over the "Color" class or whatever so that we can call "draw(shape, Cyan())", then have the actual message just hold the rgba value.

FSXAC, 2019-01-17:
Set to blocked by #183 because we need to know the form of the message before we can make a factory to build it

PR #251 (2019-01-31): "Draw logger" (Renamed to VisualizerMessenger) Implementation
### Description
Draw logger singleton used to publish shape and layer related messages.

### Testing Done
Compile passes, ran ROS node `ai_logic` locally; passed a few manual tests.

### Resolved Issues
Resolves #170

### Review Checklist
*(Please check every item to indicate your code complies with it (by changing `[ ]`->`[x]`). This will hopefully save both you and the reviewer(s) a lot of time!)*
***It is the reviewers responsibility to also make sure every item here has been covered***
- [x] **Start of document comments**: each  [... cut]
closed 2019-01-31 by FSXAC