UBC-Thunderbots/Software #176: Separate robot_communication node into a radio node and grsim node

MathewMacDougall, 2019-01-10:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The current `robot_communication` node should be separated into 2 separate nodes: 1 for radio communication, and one for grSim communication.

This will keep each type of communication separate, and allow for more specific behavior/interfaces in each.

**Describe the solution you'd like**
Separate the code that communicates with grSim into its own node, and separate the code that will deal with the radio and robot communication into another separate node.

MathewMacDougall, 2019-01-10:
If we split the nodes, we don't really have a need for the generic `Backend` class anymore, each node can just have its own class. Although both backends would take the same parameters for now, I don't see any reason why separating them would be harmful.

@garethellis0 @henrymliu @EvanMorcom @akhilveeraghanta FYI

garethellis0, 2019-01-10:
Yeah, the API for the backend is really just enforced at the ROS message level, though I wouldn't mind having them both inherit from an abstract `Backend` class (makes it easier if we ever decide to switch away from ROS).

PR #191 (2019-01-17): Separate communication nodes

## Please fill out the following before requesting review on this PR

### Description
*Give a high-level description of the changes in this PR*
Split the `robot_communication` node into separate `grsim_communication` nodes and `radio_communication` nodes

### Testing Done
*Outline any testing that was done for these changes. This could be unit tests, integration tests, etc.*
N/A
### Resolved Issues
*Link any issues that this PR resolved. Eg `resolves #1, #2, and #5` (note that they MUST be specified like this so Github can automatically close them then this PR mer [... cut]
closed 2019-01-17 by MathewMacDougall