UBC-Thunderbots/Software #178: Obtain Physical Constants MAX_ACCELERATION & MAX_ANG_ACCELERATION of the robots.

EvanMorcom, 2019-01-11:
Our robots use a bang-bang motion controller to control their motion on the field. Bang-bang controllers have 2 states: Max acceleration and Zero acceleration. For our robot firmware and software to accurately represent the motion of the robots the MAX_ACCELERATION, and MAX_ANG_ACCELERATION constants should be obtained. 

The max acceleration values can be obtained by applying maximum voltage to the robots motors, and using accelerometers/precise timing to obtain the acceleration.  

These constants can be found at: src/thunderbots/shared/constants.h

garethellis0, 2019-11-01:
@EvanMorcom @MathewMacDougall - we should properly tag this or close it. Not sure if it's worth having this ticket open right now, especially given that we're doing a new generation of robots.

MathewMacDougall, 2019-11-01:
I think we might as well close it. In our speed tests the existing values were actually kind of close to what we observed in testing.

closed 2019-11-01 by garethellis0