UBC-Thunderbots/Software #183: Determine shape messages for visualizer

qgolsteyn, 2019-01-12:


qgolsteyn, 2019-01-16:
@MathewMacDougall @garethellis0 @FSXAC Let me know what you think

garethellis0, 2019-01-16:
LGTM, could you post a what the ROS message would look like? In particular I'd like to see descriptive comments about options for `fill`, `stroke`, would be.

FSXAC, 2019-01-16:
I'm wondering how the layer and shapes messages interact. The layer message has a list of shapes, so are the shape messages sent first and then these shapes are cached?

Other minor comments/questions:
- What is the difference between `fill` and `fillShade` (likewise for the stroke data)? Wouldn't it be less work if we define `fill` to be a tuple of three RGB uint8_t values, or four RGBA uint8_t values if we want to have opacity values (likewise for stroke data).
- We already have a message type of _Point2D_ and _Circle2D_, are they being used at all? Can we reuse them here?

MathewMacDougall, 2019-01-16:
Currently `Point2D` is being used, and `Circle2D` is not.

garethellis0, 2019-01-17:
IIRC, `fill` is the type of fill (stripped, polka dots, etc.), `fillShade` is how dark the color is, `stroke` and `strokeShade` are likely similar

As to why it's structured this way, as Quentin described it to me, it's because we want to enforce the usage of a certain number of distinct colors that we have predetermined are far enough from each other so as to be visually distinguishable.

FSXAC, 2019-01-17:
Did you mean to close this?

garethellis0, 2019-01-17:
Whoops, I didn't. Good catch!

garethellis0, 2019-01-24:
@FSXAC - Not sure if you and Quentin had talked about this, but we had discussed allowing an extra degree of freedom with respect to colors. Namely, allowing us to set the shade of the color we choose ie. choose `red` but set a value from 0-255 to indicate how dark we want it? 

I think we originally had the color as being a string as well? Obviously there's a trade off between using numbers and strings, but generally we've tended to use strings to allow messages to be more readable (see the primitive message for an example).

Is there a particular reason that we don't have the stroke type in this message (just `stroke` in Quentin's initial proposal above)?

Overall though, this seem reasonable to me. I'd also like to note that this is going to require pretty robust validation on the visualizer side so we make sure we're not trying render malformed messages (and warn the user as such).

MathewMacDougall, 2019-01-24:
Overall looks good. In addition to Gareth's color comments above, I have a few clarifications:
* For `Rectangle`, I'm assuming the `x, y` coordinate is the "bottom-left" of the triangle? This should be clarified
* For the `transform_rotation`, this is the number of radians to rotate the shape? (positive being clockwise and negative being counterclockwise)? This should be clarified.
* What are the use cases for the different `types` in the Layer Message? Will this affect the coordinate transforms applied to the shapes? Where else other than the "scene" (the part that shows the field with robots) are we giving the ability to draw?

FSXAC, 2019-01-24:
RE @garethellis0:
- @qgolsteyn and I talked on Saturday and since the front end is not that far into development, it's fairly flexible.
- It wouldn't be too difficult to re-add them and change the data type for the colors, so I will update the message type to do so.
- Note that I understand we want a working visualizer and right now I want to close this issue and #170 within the next couple of days, which is why I initially opted out from extra things like shades.

RE @MathewMacDougall:
- For rectangles, the `x, y` corresponds to the top left of the rectangle, most standard graphics libraries use this convention.
  - Future improvement could be to have an option to toggle between (x, y) being in the top-left corner, or being at the center of the rectangle (perhaps similar to [Processing](https://processing.org/reference/rectMode_.html))
- Rotation is clockwise if angle is positive, and counter-clockwise if the angle is negative. Rotation is pivoting on the shape origin. (top-left corner of the rectangle, center of ellipse and arc, first vertex of polygon and line)
- I'm not sure about types as this is more of how front-end draws them. @qgolsteyn could probably answer this one

---

_I will probably create a wiki page for the message documentation and that will contain all the details listed here_

MathewMacDougall, 2019-01-24:
@FSXAC thanks for the clarification. One more I just thought of:
* What is `theta_zero` for shapes? Eg. for an arc going from theta 0 - pi/2, where does that arc start and end?

It would probably be good if the theta parameters matched the conventions used in the AI. From the "birds eye view" of the field pictured as an xy-plane, the "zero" angle is along the positive x-axis, and positive angles move counterclockwise. The Singleton could also apply coordinate transforms before sending to the Visualizer.

FSXAC, 2019-01-24:
That's correct. From the screen perspective, an arc from 0 to half pi would be a quarter-pie arc from right side to the top side (counter-clockwise).

I think the transformations should be done on the render-side. Most drawing libraries have these capabilities built-in so it would be easier.

MathewMacDougall, 2019-01-24:
Sure, that makes sense. Are the arc angles our convention, or is that standard? Seems odd to have the theta conventions for arcs opposite to the rotation transform.

FSXAC, 2019-01-24:
No, the our arc convention is reversed compared to the standard.

MathewMacDougall, 2019-01-24:
I think for consistency the rotation convention should match the arc convention.

FSXAC, 2019-01-24:
Maybe we should go with the standard convention because the front end may be using standard convention. So perhaps a bool to use field convention for the angle could be a parameter to the drawing function

FSXAC, 2019-01-24:
Ok, I've added the wiki page for the message definitions: [https://wiki.ubcthunderbots.ca/projects/software/corner-kick-visualizer/ai-drawing-messages](https://wiki.ubcthunderbots.ca/projects/software/corner-kick-visualizer/ai-drawing-messages)

Some edits:
- Changed color to be using strings
- Updated shape data definitions to be more concise and included diagrams

garethellis0, 2019-01-24:
Nice job with the documentation. Very clear! Could we put the majority of the documentation in the message itself though? In my experience there is an exponential relationship between how far away documentation is from what it is documenting, and how fast it goes out of date. In this case keeping the documentation in the message itself would mean that if anyone were to change the message, they would be _far_ more likely to also update the documentation (and it would be much more likely for it to be caught in code review.

garethellis0, 2019-01-24:
> Maybe we should go with the standard convention because the front end may be using standard convention. So perhaps a bool to use field convention for the angle could be a parameter to the drawing function

In general I think it's important that we are consistent in the coordinates that we use, but I think adding a boolean flag will just complicate things. Since the visualizer is responsible for deciding how coordinates are translated to a visual representation on the screen, could we just have it be responsible for converting coordinates, and assume that all incoming coordinates/angles are in field convention? The other option would be to have our C++ "drawing" singleton be responsible for doing the coordinate transforms before sending the message, but this seems less appropriate?

garethellis0, 2019-01-25:
Sounds good on the angles, but I'm still a bit concerned about documentation. I think that ideally the code should hold everything needed to use and understand it, and the wiki should hold high level design decisions. Applying that here the wiki would have a high level model of the drawing architecture (that the AI is responsible for creating shapes, how shapes and layers interact with the visualizer, etc) and the code would hold all the details like how different shapes are specified. I'm still not convinced that the git book will stay updated unless people are really diligent about it, which I think is a poor thing to rely on?

qgolsteyn, 2019-01-25:
Yup I agree, but not urgent at this point. I think we'll move towards in code documentation (like design notes in markdown format) as the project develops. For now, this is not critical. I think @FSXAC you have enough information to proceed on the singleton. We can always revisit this issue when we have something implemented.

qgolsteyn, 2019-01-25:
Closing this issue for now in the interest of time.

MathewMacDougall, 2019-01-25:
I believe this ticket was for discussion @garethellis0 , but the title does say implement. @qgolsteyn ?

garethellis0, 2019-01-27:
#248 resolves this, so I'm going to re-open it until that's merged.