UBC-Thunderbots/Software #195: Robot Passing

MathewMacDougall, 2019-01-12:
In order to be competitive at competition this year, we will need to re-implement our passing capabilities from the old codebase from last year.

Passing is the ability for one of our robots to pass the ball to another robot, and for that robot to successfully receive it.

The reason this Epic is estimated to be so difficult is because:
* It involves digging through a lot of legacy code
* We will need to find and extract the important parts of the passing system from last year
* These parts will need to be modified/re-written to work with the new code and architecture
* We will need to make sure the physical robots are still able to perform as we need them too
* The new logic will need to be tested and tweaked until it works

A lot of this is very unknown and likely involves many challenges we haven't predicted yet.

garethellis0, 2019-02-19:
@MathewMacDougall @qgolsteyn - not sure if we should take away points from this epic or not, now that the constituent tickets have points?

MathewMacDougall, 2019-02-20:
Yea might as well. Sub-tickets will give a better overall estimate.

closed 2019-07-24 by MathewMacDougall