UBC-Thunderbots/Software #211: Change getExtraBits() to uint8_t

henrymliu, 2019-01-12:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
The radio and firmware expects the `extra` flags of primitives to be encoded in a single `uint8_t`. In the new AI, this was changed to an array of booleans in the hope that it would be futureproof.
Unfortunately this array is time-consuming and confusing to construct, and the endianess of its contents presents an ambiguity. In addition, all current primitives either do not use this array entirely, or expect it to be either 0 or 1.

**Describe the solution you'd like**
Change `getExtraBits()` to return `uint8_t` instead of `array<bool>`, and all primitives.

**Describe alternatives you've considered**
The other alternative would be to stick to the status quo.

**Additional context**
Add any other context or screenshots about the feature request here.

FSXAC, 2019-01-12:
I got this

garethellis0, 2019-01-17:
Sorry for not seeing this sooner.

I'm pretty against this. I think it's putting too much onus on the Primitive class, making it less backend agnostic, and loses semantic meaning. When we implement the radio visitor as part of #220, I believe that should have the responsibility of packing the extra bits into a number.

FSXAC, 2019-01-17:
I'm having a difficult time seeing how would this be different except that it's a bit less clear to the programmers. Perhaps this is a non-issue?

garethellis0, 2019-01-17:
The main problem here is that it's trickier to use, easier to make mistakes, and harder to debug. Phrasing this the other way, what's the argument for changing this function instead of putting it in the "radio visitor" class? It seems like this would be less work (one function vs changing all "get Extra Bits" functions in primitives) and semantically more appropriate.

FSXAC, 2019-01-17:
I can't say that it's trickier to use. We could typedef the uint8_t to `PrimitiveFlags` as these are used exclusively for primitives. Then the flags would be assigned by bitwise OR-ing with predefined constants.

I agree that the a visitor to the primitive classes is less work. However that seems like a separate issue. If I understood correctly, your suggestion involves the inter-class interaction, specifically the `getExtraBits()` method. 

From my understanding, the objective of this task is to mostly remove the ambiguity from a variable-sized array.

What are your thoughts @henrymliu?

MathewMacDougall, 2019-01-17:
If the only concern was removing the ambiguity from a variable-sized array, one option would be an `std::array` with a predefined size of 8. This is kind of a compromise between something of a fixed size that should be easily translatable to a `uint8_t`, but an "easier" structure to work with than a raw `uint8_t` with bit-shifting (at lease on the side of the `Primitive` class)

garethellis0, 2019-01-17:
As far as I can see, the ambiguity is with the "endianess of the contents" of the array. I'm confused by this though. The array has no endianess, it's an array. The endianess should be decided by whatever is packing the bytes. Another issue pointed out is that the array is "time consuming" to construct. I'm not sure if this refers to computational time or programmer time, but I don't believe that either of those are a problem. 

@FSXAC - are you trying to say that bitwise Or-ing by constants is as easy or easy then using an array? If so, I strongly disagree, that seems more error prone and harder to debug.

Also, the radio visitor will exist either way, my point was about where the responsibility of converting a list of flags to a number should lie (I believe it should be in the radio visitor)

FSXAC, 2019-01-17:
I will try to make what I mean more clearer with an example: consider three arbitrary flags and two arbitrary primitives wants to use the extra bits.  

Primitive 1 wants to use flag A and flag B. Primitive 2 wants to use flag B and flag C.

**Using arrays**:
We insert the flag state one at a time for each primitive using a loop. The resulting arrays for both primitive extra bits arrays are:
Primitive 1: [true, true, false]
Primitive 2: [false, true, true]
Notice that there is no way to tell which index corresponds to which flag. Also note that because these arrays would have to be converted to an sequence of bits, it's unclear to see if an array like [true, true, false] would be translated to 110 or 011.

**Using uints:**
Flags are defined as:
FLAG_A = 1;
FLAG_B = 1 << 1;
FLAG_C = 1 << 2;
So to assign the extra bits, we just need to bitwise the desired flags together. For primitive 1, it would be something like `p1.extrabits = FLAG_A | FLAG_B` and for primitive 2 it would be `p2.extrabits = FLAG_B | FLAG_C`. The resulting uint for both are:
Primitive 1: 0b011
Primitive 2: 0b110
Notice the "endianess" are defined by the flag constants and no more conversion required. Use bitwise AND to access the state of flags.

As for the radio visitor responsibility, I don't disagree. But that's not in the code right now and we can do that when the time comes.

MathewMacDougall, 2019-01-17:
I do like the explicitness and readability of which flags are being set in this example.

However in your example it seems like you are assuming (correct me if I'm wrong) that different Primitives will use the same flags / flags are applicable to all Primitives. This is not the case. `extrabits = 0b001` indicates different things for different Primitives. This means each `Primitive` class would likely need to define their own flags, not that this is necessarily a problem.

@henrymliu can you clarify exactly what else is in these 8 bits in terms of the radio protocol? I know the reason for restricting to a `uint8_t` is because that's what the radio protocol uses, and we're not messing with that (which is totally reasonable). IIRC there are "generic" bits that enable autokick, enable chipping, and choose between 8 dribbler speeds that apply to all Primitives, but I'm not sure if they fall under the "extra bits" we're referring to here.

What if we left everything to do with encoding Primitives into radio packets to the `radio visitor` (or whatever)? This way the `Primitive` classes wouldn't need to worry about endianness or ordering, just the visitor code that translates them to radio packets could use the getter functions to construct the packet. However, removing the generic `getParameters` and `getExtraBits` functions means we can't use the generic `Primitive message` type to send Primitives over ROS topics.

garethellis0, 2019-01-17:
Just to make sure we're on the same page:
- the position of a flag in the array has no inherent meaning (other then what is defined in the `getExtraBits` javadoc comment for the primitive in question). So the flag at index 1 for a given primitive could be totally different then the flag at index 1 for another primitive, we do not have a universal set of flags that all primitives return (ie. MovePrimitive might have two flags, PivotPrimitive might have 1 flag, and none of those flags need be the same)

I'm not sure what you mean in your "using arrays" section?  What would you be looping over and where would this loop be? You're absolutely right that it's unclear what the array should be serialized to. This is entirely correct, as the responsibility of interpreting what the correct serialization and de-serialization of the extra bits should be is the responsibility of the radio serialization/de-serialization.

henrymliu, 2019-01-17:
@MathewMacDougall the 8 bits are primitive-dependent and are up to the firmware implementation of each primitive to interpret its contents; there are no generic bits (I think the stuff you may be referring to is encoded separately, outside of the extra bits). The MRFBackend does not care about its contents.

@garethellis0 I do not see how the current array is any easier to use for the programmer, it has made it more difficult for me to explain how to format it for new members. Primitives like Shoot use only one bit, as the entire flag is either 0 or 1. In an array form it is now {false, false, false, ..., true}, which IMO is not nice nor intuitive. A simple 0 and 1 clearly communicates "I don't want to chip" or "I want to chip".
When I say "endianness" i meant how it gets interpreted by the programmer when converting this array for use over radio (right to left vs left to right). Since I started writing this comment I see you have already acknowledged this.

@FSXAC has summarized the flag usage very well :+1: how does one immediately know which index of the array corresponds to what desired flag? If a primitive were to ever use multiple bits for flags, it would be much easier and readable to bitwise-OR flags together. I also do not see how it is easier to debug the contents of the array compared to a single int, especially if the exact bit layout of both is known, and the flags are defined constants.

garethellis0, 2019-01-17:
Ok, after some discussion with Mathew, since we should be using visitors to serialize theses primitives (either for a ROS message or radio message), I'm going to implement a visitor to serialize them to ROS messages. This visitor will just use getters on each primitive class, and so I'll also remove `getParameters` and `getExtraBits`. I believe this will keep the primitives cleaner and alleviate all this confusion.

MathewMacDougall, 2019-01-18:
Code example?:
```
void visit(StopPrimitive stop_primitive) {
    uint64_t encoded_radio_packet;
    // Do some stuff
    uint8_t extrabits = 0;
    if(stop_primitive.shouldCoast()) {
        extrabits |= 1 << 0;
    }
    if(stop_primitive.getSomeOtherThings()) {
        extrabits |= 1 << 1;
    }
}
```

henrymliu, 2019-01-18:
Okay, I think I have some gaps in knowledge on what you two are planning. We can discuss on Saturday.

FSXAC, 2019-01-24:
Closing this issue for #235

closed 2019-01-24 by FSXAC