UBC-Thunderbots/Software #228: Normalize the timestamping system

MathewMacDougall, 2019-01-19:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
After the change in #179 we now have several timestamping systems in use. We should normalize our useage of timestamps to a single system.

Depends on #227 since managing timestamps will be easier after the nodes are callback-based.

**Describe the solution you'd like**
We should typedef `double` to `Timestamp`, and use this as the timestamp standard. Remove the old AITimestamp class (that wrapped std::chrono), and remove our use of std::chrono.

**Describe alternatives you've considered**
Using a different timestamping system like std::chrono. We should use the t_capture values so everything is as close to the "real world" timestamp as possible.

**Additional context**
None

MathewMacDougall, 2019-01-23:
One thing I just realized is that the `ssl game controller` (the replacement for refbox) uses a different timestamping system than `SSL Vision`. Rather than defining a time in seconds since the "epoch" of when the application started, the SSL Game Controller uses a UNIX timestamp, in microseconds. The `.proto` file is [can be found here](https://github.com/RoboCup-SSL/ssl-game-controller/blob/master/pkg/refproto/ssl_referee.proto).

This may pose problems if we need to synchronize SSL Vision data or internal state with the Game Controller Commands. I don't think this is _currently_ an issue, as our `GameState` class doesn't use the timestamp (even though it's in the ROS message). @jordanzliu FYI.

I'm going to try take a look at what RoboTeam Twente did. I suspect they just don't deal with the Game Controller timestamp (as our `GameState` is based off their code).

@henrymliu @garethellis0 FYI

garethellis0, 2019-01-23:
I suspect that your assumption is correct. It makes a lot more sense to base decisions off of the timestamps of the data that is the primary decision driver.

PR #274 (2019-02-03): Normalizing timestamps to use new Timestamp class

## Please fill out the following before requesting review on this PR

### Description
Replaces the previous timestamping system(s) with a single, unified `Timestamp` class. This new class is built with the `t_capture` timestamp we receive from protobuf packets in mind, and should be simpler to work with than `std::chrono`

### Testing Done

* Unit tests for the new Timestamp class
* All other unit tests have been updated to use the new Timestamp class

### Resolved Issues

resolves #228 

### Length Justification
Once I modified the old `AITim [... cut]
commit a917e024 (2019-02-03): Normalizing timestamps to use new Timestamp class (#274)

## Please fill out the following before requesting review on this PR

### Description
Replaces the previous timestamping system(s) with a single, unified `Timestamp` class. This new class is built with the `t_capture` timestamp we receive from protobuf packets in mind, and should be simpler to work with than `std::chrono`

### Testing Done

* Unit tests for the new Timestamp class
* All other unit tests have been updated to use the new Timestamp class

### Resolved Issues

resolves #228 

### Length Justification
Once I modified the old [... cut]
closed 2019-02-03 by MathewMacDougall