UBC-Thunderbots/Software #232: Put the shared constants in a namespace

MathewMacDougall, 2019-01-19:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
The shared constants in `shared/constants.h` should be put in a namespace to match the constants in `software/util/constants.h`. Because this file will also be used by the C code in the robot firmware, the namespaces will need to be wrapped in `#ifdef` statements so they are only compiled in with the C++ code.

**Describe the solution you'd like**
Add namespaces to the `shared/constants.h` constants, and wrap them in `#ifdef` statements so they are only used for C++

**Describe alternatives you've considered**
None

**Additional context**
None

garethellis0, 2019-11-20:
This is very out-of-date now, @MathewMacDougall can we close this?

MathewMacDougall, 2019-11-20:
Is this not something we would still do? It might be good to consolidate our constants

garethellis0, 2019-11-21:
Right, but I don't think the namespacing bit is relevant anymore?

MathewMacDougall, 2019-11-21:
Ah you're right, this can be closed

closed 2019-11-21 by MathewMacDougall