UBC-Thunderbots/Software #255: Build the logic for the console

qgolsteyn, 2019-01-26:
### Description of the task

Add logic to receive and log messages from /rosout

### Outline the task in more detail

This work is needed to implement the console of corner_kick. We should add a new saga and reducer called `console` which receives messages from /rosout using the actions defined in `actions/ros.ts`

### Acceptance criteria

- [ ] Subscribes to /rosout
- [ ] Receives messages from /rosout and adds them to the Redux state

qgolsteyn, 2019-01-29:
You will need to create a new reducer and saga object with a separate state.

qgolsteyn, 2019-01-29:
Essentially, you want the saga to subscribe to the rosout topic and the reducer to add messages to the state when receiving actions of type ros_NEW_MESSAGE with ros topic /rosout.

MathewMacDougall, 2019-01-29:
Just to clarify, where is this going to show up in the Visualizer and what is the purpose? This is different from the `Annunciator/Robot status` messages right?

`/rosout` is going to contain *_ALL_* messages published to the system, which includes all ROS messages (`Ball`, `Team`, `Primitives`, etc.) AND all output from our `LOG` statements. Unless some super-fancy filtering is done, the `Ball`, `Team` and similar messages will drown out anything useful (like the LOG calls).

I think it might be preferable to publish `LOG` message to their own topic and just display those? This can still be separate from the `ROBOT_STATUS` messages.

PR #277 (2019-02-02): Implement console logic

### Description

The console will require its own logic to process messages from the /rosout topic. This will be done with the saga subscribing to /rosout and the reducer will send the messages to the state.

### Testing Done

Unit tests for the reducer

### Resolved Issues

Resolves #255 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Start of document comments**: each `.cpp` and `.h` file should have a comment at the start of it. See files in the `thunderbots/softwa [... cut]
closed 2019-02-02 by kelvinkoon