UBC-Thunderbots/Software #268: Create a General EnumString Class

garethellis0, 2019-01-27:
**Is your feature request or enhancement related to a problem? What specifically could be better? Please describe.**
Create a `EnumString` class that allows for us to create a compile-time enum with corresponding methods to lookup the enum values to strings (ie. with macros and/or templating). This will be important in allowing us to interface to the visualizer, which runs primarily off strings.

This will be a generalization of the enum-string conversion stuff added in #251.

Note: name of this class is poor and should be changed.

**Describe the solution you'd like**
See above.

**Describe alternatives you've considered**
Runtime enum to string conversion. Ideally we want compile time checks, so this is out.

**Additional context**
NA

garethellis0, 2019-01-27:
The first answer on https://stackoverflow.com/questions/28828957/enum-to-string-in-modern-c11-c14-c17-and-future-c20 is probably what we want.

MathewMacDougall, 2019-10-04:
Replaced by #857

closed 2019-10-04 by MathewMacDougall