UBC-Thunderbots/Software #318: Cleanup The CMakeLists.txt

garethellis0, 2019-02-03:
### Description of the task
Cleanup the `CMakeLists.txt` with a focus on reducing it's length by reducing the amount of duplication present.

### Outline the task in more detail

Currently the `CMakeLists.txt` is over 580 lines. Once #217 is completed we should be able to remove the inclusion of _every_ primitive in every that uses _any_ of the primitives (including the tests). Additionally, we should make a variable with the common libraries we use (`catkin_LIBRARIES` and `G3LOG` at the time of this writing) so that we don't duplicate the inclusion of them everywhere. We can also remove all inclusions of headers directly in the target. Those were originally there because CLion needed them in the target in order to find them, but this issue was resolved 8+ months ago, so everyone should have a CLion version that now can just find the headers without needing them explicitly declared in the target.

### Acceptance criteria

- [ ] Consolidate things into variables if possible, in particular commonly used libraries
- [ ] Remove any unneeded inclusions of primitives in targets (probably mostly in test targets) 
- [ ] Remove all inclusions of headers in targets

MathewMacDougall, 2019-02-03:
We should also be able to remove any headers included  in the executables. These were included before in order for CLion to properly treat them as project files, but this has been fixed in later versions.

garethellis0, 2019-02-03:
Yup, this is covered in #19, mentioned above.

MathewMacDougall, 2019-02-03:
Whoops didn't see that.

MathewMacDougall, 2019-10-06:
No longer relevant with bazel

closed 2019-10-06 by MathewMacDougall