UBC-Thunderbots/Software #352: Move old firmware to new repo

Thunderbots, 2019-02-16:
### Description of the task

Move the robot firmware from the old GitLab repo to our new GitHub repo

### Outline the task in more detail

We need to move the robot firmware as-is to the new github repository. It should go in `src/thunderbots/firmware`. The `setup_firmware.sh` script will need to be added to our `environment_setup` folder. Most importantly, the CMakeLists.txt file will need to be moved and confirmed that is still works. It uses commands to flash the robot firmware.

### Acceptance criteria

- [ ] firmware is moved to `src/thunderbots/firmware`
- [ ] firmware setup script is moved to `environment_setup`
- [ ] Code builds and robots can be flashed

garethellis0, 2019-03-03:
@MathewMacDougall - is this gonna get done? I'm taking it off the milestone for now.

MathewMacDougall, 2019-03-03:
Not my highest priority this week, so that's fine.

PR #487 (2019-03-16): Move old firmware to new repo

## Please fill out the following before requesting review on this PR

### Description

Moved the firmware from the old repo to our current one. Added a few scripts to make flashing easier, and added basic instructions.

### Testing Done

I am able to build the firmware code, and the firmware unit tests pass.

I confirmed I can flash both the robots and dongle, and that they still work afterwards.

### Resolved Issues

resolves #352 

### Length Justification
commit 0903eea0 (2019-03-16): Move old firmware to new repo (#487)

## Please fill out the following before requesting review on this PR

### Description

Moved the firmware from the old repo to our current one. Added a few scripts to make flashing easier, and added basic instructions.

### Testing Done

I am able to build the firmware code, and the firmware unit tests pass.

I confirmed I can flash both the robots and dongle, and that they still work afterwards.

### Resolved Issues

resolves #352 

### Length Justification
closed 2019-03-16 by MathewMacDougall