UBC-Thunderbots/Software #382: Implement Pass Generation For PassGenerator

garethellis0, 2019-02-21:
### Description of the task
Fully implement the `generatePasses` function in the `PassGenerator` class.

### Outline the task in more detail
We should roughly base this function on how pass generation was done in the old code. See https://wiki.ubcthunderbots.ca/-LNDlFcpV2p--IwnWNlJ/software/passing#advice-sidenotes-random-info for details.

### Acceptance criteria

- [ ] `generatePasses` in `PassGenerator` is implemented

garethellis0, 2019-02-23:
@JamesPetrie - could you give a quick overview of the impetus behind re-generating passes? Are we trying to re-generate passes around current passes to help avoid local minima? Are we generating passes as far away as possible to make sure we're exploring the entire world as it changes? Or something else entirely?

JamesPetrie, 2019-02-24:
A little bit of both really. The ideal case is that there are ~10 local maxima and each of them is being tracked as the world changes. Because local maxima could quickly change to global maxima, and also to give alternative pass options.

In order to explore as much of the space as possible we wanted to quassi-randomly test new starting points. Since points are being removed by the merging they have to be added back anyways. And the worst points are removed under the assumption that if they were close to a good point 10 GD iterations would be enough to make them better than the bottom 20% of passes.

The logic used in the old code actually didn't consider the location of current points when generating new ones, but this was mainly because that made it less complicated.

We initially considered uniformly sampling from the space, but chose a heuristic to give a bit more direction to the search.

To do this the field is considered in radial coordinates with the centre at the passer position. It is assumed that each enemy causes a local minima in the angle it is at, and each friendly robot causes a local maxima at its radius. So the 'best' angles are the ones directly between enemy robots (at the midpoint) and the best radiuses are the ones our robots are at.

This gives roughly 6*4 potential starting points (6 enemies = 6 angular midpoints, 6-2 potential passees gives 4 radiuses).

I think the positions were sampled without replacement from these choices, and if more were needed they were sampled uniformly from the field. I believe the guess for ball speed and delay time was just hard coded, but it may be good to add some randomness to this too.

Kind of hard to describe with words so let me know if a diagram would be useful

garethellis0, 2019-02-24:
Awesome! That makes sense, but could I get some clarification or a diagram on what you mean by each enemy causing a local minima in the angle it is at?

JamesPetrie, 2019-02-24:
[Pass Generation Heuristic.pdf](https://github.com/UBC-Thunderbots/Software/files/2897583/Pass.Generation.Heuristic.pdf)

What I mean by minima is that it usually isn't good to pass in the direction of an enemy, so test angles that are as far from enemies as possible. (Enemies create 'valleys' behind them in the pass quality function).

Now even from this diagram it shows that this heuristic is missing some potential areas. It might be good to add in some phantom enemies every 60 degrees or so if there aren't any enemies there so we don't completely miss areas of the field.

garethellis0, 2019-02-24:
Ah, makes sense, thanks!

MathewMacDougall, 2019-02-24:
Interesting heuristic, and nice image (very useful) :+1:

JamesPetrie, 2019-02-24:
Just double checked and it looks like half of the generated points used this heuristic and the other half were sampled uniformly (using the wrong field dimensions though). And for the heuristic it looks like we tried to guess a good ball speed based on how fast to kick to avoid one of the closer enemies intercepting, and then guessed the delay time based on how long it would take our robot to get there and the given ball speed.

Now this may all have been overkill. I vaguely remember testing without the heuristic and it taking longer to converge to reasonable pass points. If the goal is to get to minimum viable product as fast as possible it may be worth leaving this function just sampling uniformly and coming back to it after the rest of the passing stuff is done.

garethellis0, 2019-02-24:
Excellent. Thanks for the guidance on this James, that's super helpful.

PR #491 (2019-03-19): Implemented Pass Generation

## Please fill out the following before requesting review on this PR

### Description
Implemented uniform pass generation in the pass generator. Also migrated `ratePass` to the `evaluation.cpp` file so we can independently test it and added unit tests for it.

### Testing Done
Basic unit tests. More will likely be needed when we get to actually tuning the pass generation.

### Resolved Issues
resolves #382

### Length Justification
NA
closed 2019-03-19 by garethellis0