UBC-Thunderbots/Software #383: Implement ratePass In PassGenerator

garethellis0, 2019-02-21:
### Description of the task
Implement the `ratePass` function in the `PassGenerator` class.

### Outline the task in more detail
We should roughly base the implementation of this on how it was done in the old code. See here for details: https://wiki.ubcthunderbots.ca/-LNDlFcpV2p--IwnWNlJ/software/passing

### Acceptance criteria

- [ ] `ratePass` in `PassGenerator` is implemented

garethellis0, 2019-03-01:
@JamesPetrie - what was the logic behind rating risk based on both the enemy intercept risk and the distance of the enemy to the receive point of the pass? (in `enemy_risk.cpp` in the old codebase)

JamesPetrie, 2019-03-01:
I think the idea with that was that there were two separate considerations: the probability of the ball getting to the passee, and the amount of space the passee has once the ball arrives.

This sort of distinguishes between the case where the enemy robot is really close to the passer, and the case where the enemy robot is close to the passee. Both might have an equally likely chance of interception, but you'd prefer the one where the location you're passing to is empty of enemies.

garethellis0, 2019-03-01:
Ah perfect. Thanks for the context James!

closed 2019-03-09 by garethellis0