UBC-Thunderbots/Software #388: Tune The PassGenerator

garethellis0, 2019-02-23:
### Description of the task
Tune the various hyperparameters that the `PassGenerator` uses to converge to good passes.

### Outline the task in more detail
Look into tuning the following in the `PassGenerator`:
- param weights
- hyperparameters for the gradient descent optimizer that we use (see the class constructor for the gradient descent class)
- all the dynamic parameters used by the `PassGenerator`

The goal is to maximize convergence to good passes in game scenarios, so this will likely involve some work setting up situations in grsim to test.

### Acceptance criteria

- [ ] Aforementioned hyperparameters are tuned as best as possible

MathewMacDougall, 2019-06-19:
@garethellis0 I think we can consider this complete?

garethellis0, 2019-06-19:
Yup, closing.

closed 2019-06-19 by garethellis0