UBC-Thunderbots/Software #390: Modify The VisualizerMessenger To Allow For Websocket Communication

garethellis0, 2019-02-24:
### Description of the task
Update the `VisualizerMessenger` class so that it sends messages to the visualizer over a websocket.

### Outline the task in more detail
We essentially need to make the `VisualizerMessenger` a websocket server that can serve shapes to the visualizer over a websocket. I've written a really hacky version that can send over a rectangle, but we should do the whole thing cleanly.

For the current work done, please see the `garethellis0` fork, branch `gareth/websocket-testing`, or see the attached files.

Please note, this is dependent on some changes on the visualizer side of things as well, please see the `qgolsteyn` fork, branch `feature/workers`, for all the relevent changes there.

### Acceptance criteria

- [ ] Modify the VisualizerMessenger to send shapes via websocket

[visualizer_messenger.zip](https://github.com/UBC-Thunderbots/Software/files/2897672/visualizer_messenger.zip)

qgolsteyn, 2019-02-25:
# Design Notes for the Visualizer Canvas communication protocol

The visualizer canvas is an area in the visualizer interface that offers the ability to render arbitrary shapes, organized in layers. To increase performance, the appearance of each shape must be defined at startup. We refer to appearance of a shape as a texture. Each instance of a shape can modify the position, dimension, rotation and color of the shape's texture in real-time.

This document refers to the process by which shape data can be transmitted to the visualizer canvas. The visualizer canvas will abide to this process, but may optimize its internal data flow to increase performance.

## Glossary
_Layer_: A layer is a structure containing an arbitrary number of shapes. They are identified with a name and can be reordered within the visualizer canvas.
_Shape_: A unit of graphical output in the visualizer. Displays a arbitrary texture in the canvas, and allows for graphical operations to be performed on said texture.
_Texture_: An arbitrary shape. All textures must be generated at startup.

## Overview of the communication protocol between the visualizer canvas and the AI
The visualizer canvas expects two ROS services to be available at runtime. The first, called the TextureService, should provide a list of all textures that may be requested in the visualizer canvas. The second, called the LayerService, should provide layer information to the visualizer canvas, including layer name and maximum shape count.

Shape information will be transmitted to the visualizer canvas in a binary format over a WebSocket. The visualizer canvas will initiate shape transmition by sending a message to the AI, when performance allows for messages to be received. The AI will respond by sending shapes for each layer, with one message per layer. Alternatively, to improve performance, the AI may respond with a no update required if no shapes have been changed for a particular layer since the last time shape information was requested.

qgolsteyn, 2019-02-25:
Details of the data format to follow shortly

qgolsteyn, 2019-02-26:
Data format for shapes updated

MathewMacDougall, 2019-02-26:
>>>
The visualizer canvas will initiate shape transmition by sending a message to the AI, when performance allows for messages to be received.
>>>
Why not expose this as a service instead, and just have the Visualizer request the latest data? This seems much more appropriate than listening for a message and publishing something once it's received.

garethellis0, 2019-02-27:
> Why not expose this as a service instead, and just have the Visualizer request the latest data? This seems much more appropriate than listening for a message and publishing something once it's [...]

I imagine the by "sending a message", we mean over the websocket here? We can't transmit anything with a significant performance requirement (like what to draw at any given point) over anything in ROS, because the ROSBridge/roslibjs side of things is simply too slow (it serializes everything to JSON, which then needs to be parsed by the visualizer).

garethellis0, 2019-02-27:
@qgolsteyn - could we formalize the intermediate spec we talked about a bit over the last few days? As I understand it, it will basically be the above spec, but without:
- TextureService or LayerService
- Shape update requests (ie. the visualizer will just send over all the layers once per tick without being asked to)

qgolsteyn, 2019-03-17:
Couple of edits required:

### Change `drawRect`, `drawEllipse`, `drawLine` to `drawSprite`
Because the visualizer uses predetermined textures, there is no such thing as drawing a `rect`, `ellipse`, `line` anymore. Rather, everything is a sprite, and you can draw certain shapes by specifying the correct texture in the visualizer's spritesheet (still needs to be implemented).

### Do not restrict layer creation
I understand in the previous implementation we limited the number of layers. This is not necessary for this implementation. We support adding shapes to 256 unique layers identified by their layer number. We should support adding any shapes to any layers within that range instead of throwing a warning log message.

### Support client disconnect
When developing corner_kick, we reload the app everytime the code changes. This disconnects us from the websocket, but the server still sends messages to the closed connection.

### Properly handle process termination
Properly terminate any threads and disconnect from the socket when the ai_logic process is terminated.

### Send shape messages in big endian
TCP uses big endian as a standard when sending messages between server and browser. Right now, the websocket sends binary data in little endian and the browser expects big endian. We need to change the endianness of the server to respect this standard.

qgolsteyn, 2019-03-17:
@FSXAC @garethellis0

FSXAC, 2019-03-20:
Can you split that into five different issues? I don't know about the first one though. Sounds like extra work for the end user for pretty much no change on the visualizer side.

garethellis0, 2019-03-20:
Yeah, I know we talked about this @qgolsteyn, but I don't think I understood correctly. I think having drawRect, drawEllipse, etc. is a better approach for a few reasons. It gives us a compile time guarantee that we're trying to draw something valid, and ensures that the c++ drawing api is limited enough that we can switch over to drawing some other way later without having to change all the "draw" calls. I also agree with @FSXAC that it's a nicer user experience.

TL;DR: visualizer messenger is supposed to abstract all the spritesheet stuff away and should not leak through this abstraction.

qgolsteyn, 2019-03-21:
Is it? I don't think it necessarily has to abstract that away. What if you want to move a robot? Is that drawing a shape or moving a sprite? Personally, I think it makes more sense shape drawing occurs when generating the spritesheet (which could happen on the AI at some point) and during runtime you are moving predrawn shapes.

Feel free to disregard my opinion though. As long as the message you are sending is valid and takes into account the constraints of the canvas (predefined texture meaning additional logic on the AI side to give the impression of "drawing" shapes), then I'm totally cool with whathever you come up that fits your requirements.

garethellis0, 2019-03-21:
I understand your point here, but I still feel pretty strongly that how we're drawing something should be abstracted from the C++ logic side of things, so I stand by my original comment.

qgolsteyn, 2019-03-21:
Sounds good!

MathewMacDougall, 2019-04-27:
Closing as we are now using ROS to send binary data rather than a websocket

PR #445 (2019-03-16): VisualizerMessenger with Websocket
## Please fill out the following before requesting review on this PR

Uses websocket instead of ROS messages for transmission of shape data from ROS to visualizer.

### Testing Done

Compiled and rosrun'd. Still require integration testing on the visualizer/receiving end of the websocket.

### Resolved Issues

Resolves #390

### Length Justification

N/a

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Start of document comments**: each `.cpp` and `.h` file should have a comment at the [... cut]
closed 2019-04-27 by MathewMacDougall