UBC-Thunderbots/Software #398: Clean Up geom/util

garethellis0, 2019-02-27:
### Description of the task
Perform a few small cleanup actions for the functions in `geom/util.[h,cpp]`.

### Outline the task in more detail
Please see the acceptance criteria  below for a list of revisions.

### Acceptance criteria

- [ ] Merge all `*intersect` functions into a set of `intersection` functions, where they just differ by the arguments they take, similar to the `intersects` functions
- [ ] Make sure similar functions are grouped together

garethellis0, 2019-02-27:
Just leaving this here for use when we do this:
```
/**
 * Finds the intersection of a given Ray and Segment, if there is one
 *
 * @param ray
 * @param segment
 *
 * @return
 *      - a tuple of (std::nullopt, std::nullopt) if there is no intersection point
 *      - a tuple of (point, std::nullopt) if there is one intersection point
 *      - a tuple of (point, point), if the ray and segment are parallel, where the two
 *        points are the end-points of the line segment made up by the overlap of the
 *        ray and line segment
 */
std::tuple<std::optional<Point>, std::optional<Point>> raySegmentIntersect(const Ray& ray, const Segment& segment);

std::tuple<std::optional<Point>, std::optional<Point>> raySegmentIntersect(const Ray& ray, const Segment& segment){
    auto isect = lineIntersection(ray.getRayStart(), ray.getDirection(),
                                  segment.getSegStart(), segment.getEnd());

    // Check that there was an intersection and it is within the ray and segment
    if (isect.has_value() && contains(ray, isect.value()) && contains(segment, isect.value())) {
        // Check if the ray and segment are parallel
        if(collinear(ray.getRayStart(), ray.getDirection(), segment.getSegStart())){
            // The start point of the overlapping segment between the ray and segment
            Point overlapping_seg_start = closestPointOnSeg(ray.getRayStart(), segment);
            
            // TODO: calc second point here
        }

        // They are not parallel, so just return the intersection
        return {isect, std::nullopt};
    }

    return {std::nullopt, std::nullopt};
}
```

jsn0716, 2019-11-05:
Now included in #959

closed 2019-11-05 by jsn0716