UBC-Thunderbots/Software #412: Automatic cpp and cfg generation for DynamicParameters

akhilveeraghanta, 2019-03-06:
### Description of the task

Currently to add a parameter, we need to edit both a cfg file and a cpp/header file. This adds a lot of steps and complexity to something that should be simple to setup and use. 

This ticket involves implementing a code generator in python to generate both the cfg/cpp files from a yaml file, where all the parameters are defined in a specified format. 

Generate cfg and cpp from yaml for DynamicParameters

### Outline the task in more detail

- Depends on DynamicParameters

### Acceptance criteria

- [ ] YAML to CPP and CFG conversion works and has been verified
- [ ] Passes param exists in param server integration test
- [ ] Supports all 4 xmlrpc types
- [ ] Supports lists of parameters of same type
- [ ] Supports multiple pages of parameters
- [ ] Making sure that the parameter updates are seen across all the nodes.
- [ ] Documentation on the wiki on the structure of the parameters
- [ ] Readme on how to use the new setup

garethellis0, 2019-03-08:
@akhilveeraghanta - is it realistic to get this in this week? I'll open the milestone for next week, but it's totally your judgement call.

akhilveeraghanta, 2019-03-08:
I finished the core of the generation and tested it all that’s left is the Singleton, I’m hoping for a PR up tonight, if something comes up I’ll let you know

PR #447 (2019-03-13): DynamicReconf: Generate cpp/cfg from yaml

## Please fill out the following before requesting review on this PR

### Description
This pull request contains the changes to create an automatic cpp/cfg generator based on the yaml file specified. 

### Testing Done
The integration test "parameter exists in server" still covers this. There is no unit testing framework currently in place for the python module but it has been tested manually

### Resolved Issues
resolves #412 

### Length Justification

This pull request mostly contains changes that involve deleting files (adding them to the gitignor [... cut]
closed 2019-03-13 by akhilveeraghanta