UBC-Thunderbots/Software #428: Implement Catch Action

MathewMacDougall, 2019-03-09:
TODO

henrymliu, 2019-04-20:
I think the description is a bit wonky; "...kick the ball at a certain speed"

Anyways, this seems like an interesting one. I'm working under the assumption that `jcatch.c` is used in firmware... which seems incomplete when I read through the file. Did this work well last year?

The old catch action is just as sketchily written and doesn't seem to follow coroutines properly, which obviously will have to be done in this new version.

henrymliu, 2019-04-27:
After some field testing, it has been determined that the firmware implementation is not working properly. Since this wasn't used in competition, I've moved this issue to the backlog pending a firmware primitive rewrite.

closed 2019-07-24 by MathewMacDougall