UBC-Thunderbots/Software #432: Implement friendlyTeamHasPossession function

MathewMacDougall, 2019-03-09:


MathewMacDougall, 2019-03-18:
@matthewberends @garethellis0 I want to start some preliminary discussion here, on exactly how to handle the case mentioned above where a team may not have immediate control of the ball (ie. the ball is in a robot's dribbler), but can still be considered to be in the team's possession, such as when the ball is traveling during a pass.

We want to handle this case well since if we use this function for the `applicable` and `invariant` conditions of a Play, we could rapidly change plays as the ball moves around even if it doesn't make sense to do so.

Evidently some kind of state will need to be store, but exactly how isn't clear. A debouncing-based solution seems best, so that the function will only change its value if the `N` most recent values are all the same. That way if the team temporarily "loses" possession while the ball is traveling for a pass, we don't immediately return false. However, it's hard to store state in the function since it's generic and can be used for different teams, but it's also non-trivial to provide it with a list of past value, since it may be used multiple times from multiple places.

Any ideas?

garethellis0, 2019-03-18:
This could be a place for a coroutine? We could hold the debouncing values in a variable in the function.

@matthewberends - not sure how familiar you are with these, so gonna leave some context here (this will eventually end up on the wiki)
Coroutines essentially mean that you start running a function, then return a value, but also return the rest of the work to be done, so that you can _keep_ running the function later on without losing state. We use these in tactics and actions as an alternative to having a giant state machine. 
The wikipedia article on this is pretty good, I would suggest reading the intro and the "Comparison with subroutines" section. Please make sure you understand this (and totally feel free to ping us on slack with questions), as it's generally a pretty important aspect of our codebase, and it will be impossible to have a discussion about this otherwise.

MathewMacDougall, 2019-03-18:
After some discussion with Gareth, here's what we came up with:

We should update the `Robot`, `Team` and `Ball` classes to store their previous `N` states (`N` would be specified in the constructor, like how we specify the robot disappearing delay in the `Team` constructor). Each class should now also implement a `getStateAtPreviousTime` function that returns the state at the previous time by interpolating the past states. It should return std::nullopt if the time requested is further in the past than the stored state.

This will make it easy to implement any hysteresis-based functions in the future.

Then we can use this to write the `teamHasPossession` function, which can then use the previous states of the ball and team to do any debouncing / hysteresis calculations. The `getStateAtPreviousTimes` functions are useful to make sure we compare values at the exact same time/state, just in case things aren't updated at _exactly_ the same time (due to different cameras, etc.)

MathewMacDougall, 2019-03-18:
Tickets #498, #499, #500 created for storing previous state

PR #601 (2019-05-26): Team Has Possession Evaluation Function

## Please fill out the following before requesting review on this PR

### Description

Evaluation function to determine that a team has possession of the ball.

### Testing Done

Unit tests

### Resolved Issues

resolves #432 

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [x] **Start of document comments**: each `.cpp` and `.h` file should
closed 2019-05-26 by matthewberends