UBC-Thunderbots/Software #438: Implement ballInOurCorner function

MathewMacDougall, 2019-03-09:


MathewMacDougall, 2019-03-20:
@garethellis0 	@matthewberends 	@HaomiaoZ 	wondering what type of distance metric we should use here to check if the ball is in the corner.

*Option 1: Euclidian distance*
* The distance parameter would then be the "radius" from the corners the ball must be within to be considered in the corner

*Option 2: Chebyshev distance*
* The distance parameter would then be the side of the square (with one corder in the field corner and the other in the field) that the ball must be within to be considered within the corner

The Euclidian distance is a bit easier to think about, and is probably what people would assume the function uses as first glance, but Chebyshev sort of handles the rectangle-ness of the field nicely.

garethellis0, 2019-03-20:
I'd go for the easier to understand option if there's no evidence that the more difficult option provides some significant benefit.

matthewberends, 2019-03-21:
Euclidian distance seems to intuitively make more sense to me. The advantage of having a square with a corner in the field corner is not evident to me though so if it does provide a major convenience it might sway my opinion.

Also I assume that this function is to assist with corner kick procedure and for that, wouldn't the ball need to be within a circle of some radius that is centred around the corner. If that is the case, it seems like Chebyshev distance would cause situations where the ball is outside of the corner kick circle but technically in the distance square and cause issues?

MathewMacDougall, 2019-03-21:
Yes this function is useful for detecting when we want to take/defend corner kicks rather than normal free kicks. I agree with Euclidean, and have updated the ticket.

@matthewberends not sure I totally understand your last point. Chebyshev distance would technically cover a larger area, but not sure about issues that would cause.

matthewberends, 2019-03-22:
@MathewMacDougall Yeah I think you are right. I did some reading on Chebyshev distance and I was misunderstanding it. Sorry about that

closed 2019-04-28 by akhilveeraghanta