UBC-Thunderbots/Software #484: Remove thread and locks from VisMessenger

FSXAC, 2019-03-14:
> Ah, so this looks like we're not joining to the thread we have spawned. Basically if you destroy an `std::thread` object that is still running, it will call `std::terminate` and just kill your program. In this case that's somewhat OK, in that we were trying to stop it anyways, but this would be a serious problem if we ever tried to destruct a `VisualizerMessenger` in the middle of execution. The fix here is basically to tell the `std::thread` object that it's time to end in the `VisualizerMessenger` destructor, and then wait for it to end before destroying the `std::thread` object. I had to deal with this for the `PassGenerator`, checkout the destructor for that here (https://github.com/UBC-Thunderbots/Software/blob/master/src/thunderbots/software/ai/passing/pass_generator.cpp).
> 
> After thinking about this some more and reading up on the boost `tcp::acceptor` documentation, we could remove all threading stuff and just use `async_accept` instead of `accept`? Here's an example: https://stackoverflow.com/questions/1087199/boostasio-how-do-i-use-async-accept-to-accept-incoming-connections. This would remove all the threading and locking issues.

_Originally posted by @garethellis0 in https://github.com/UBC-Thunderbots/Software/pull/445#issuecomment-472439874_

garethellis0, 2019-03-14:
@FSXAC - could you fill out the issue template for this issue?

FSXAC, 2019-03-14:
Yep, I'll get around to it soon

MathewMacDougall, 2019-04-27:
Closing as this isn't relevant anymore, since we're using ROS to send binary data now.

closed 2019-04-27 by MathewMacDougall