UBC-Thunderbots/Software #500: Store previous states in Team class

MathewMacDougall, 2019-03-18:
### Description of the task

Store some number of previous Team states in the Team class.

### Outline the task in more detail

The member variables of the `Team` class should now be queues or circular buffers of some size `N`, which should be defined in the constructor. The current functions can then just use the most recent value in these queues.

Additionally, new functions should be added to return a copy of the queues. Eg. `getPreviousPositions()`.

Most unit tests should stay the same and pass with the new implementation. Only a few new tests need to be added for the new `getPreviousStates` functions.

### Acceptance criteria

- [ ] Convert the member variables to queues/circular buffers of their current type
- [ ] Update existing functions to work with the queues
- [ ] Add new functions to return a copy of the queues
- [ ] Add unit tests for the new functions

MathewMacDougall, 2019-03-18:
@matthewberends 	@garethellis0 	The previous states for the `Team` is a bit weird, so we should figure out exactly what the interface should be.

Right now, the member variable the `Team` uses is a a map of robot ids to `Robots`. Each robot is now also going to be storing its previous state.

There are 2 options here:
1. The `getPreviousStates` function for the `Team` returns a vector of maps, in which case we can account for robots appearing/disappearing (although this may not be really necessary for any functions that rely on previous state. This is technically more accurate, but potentially harder to work with.
2. The `getPreviousStates` almost doesn't need to do anything since the previous robot states can be accessed from the `Robot` objects in the map.

The history of timestamps and goalie_ids are trivial. The options above are only for the robots on the team. After some thought, I think the first option isn't actually that bad and is the better solution here.

Thoughts?

garethellis0, 2019-03-18:
I think option (1) is probably the more sane option here.

MathewMacDougall, 2019-04-27:
Closing as this isn't necessary for possession evaluations (it just stores more data without real benefit)

closed 2019-04-27 by MathewMacDougall