UBC-Thunderbots/Software #516: Extra elements in calcBestShotOnEnemyGoalAll return vector

HaomiaoZ, 2019-03-24:
### Description of the task

Both `calcBestShotOnEnemyGoalAll()` would return extra elements along with other correct elements.
 
### Outline the task in more detail

- `angleSweepCirclesAll()` in `geom/util.cpp` returns extra elements to `calcBestShotOnEnemyGoalAll()` in `ai/hl/stp/evaluation/calc_best_shot.cpp`
- 2 unit tests for `ai/hl/stp/evaluation/calc_best_shot.cpp` failed because of this. Suggestion would be debugging those failed tests and fix the bug accordingly
- Currently the extra elements would occur when one obstacle block the positive post, or when one obstacle is fully below the negative post. Other cases such as multiple obstacles lower than negative post should be tested
- Problem #303 and PR #407 could help the understanding of the problem and current unit tests

### Acceptance criteria

- [ ] No extra elements in the return vector of `calcBestShotOnEnemyGoalAll()`

EvanMorcom, 2019-04-24:
@MathewMacDougall Can we close this? I believe you removed this function in another PR.

MathewMacDougall, 2019-04-25:
Ah yup you're right. Gareth also fixed the `angleSweep` implementation so the underlying problem here should be fixed too.

closed 2019-04-25 by MathewMacDougall