UBC-Thunderbots/Software #518: Read and write to ROS Param Server using Visualizer UI

kelvinkoon, 2019-03-25:
### Description of the task
Read the ROS Param Server's data by converting the YAML files's information to JSON and hydrate it into the state. 
Write to the ROS Param Server using the Service Request.

### Outline the task in more detail
The state should store the name, value, description, range, and type. The saga should fetch the JSON, hydrate the state (fetch the current value), and update the value (validate the value, update the ROS param server using the Service Request) and update the state.

### Acceptance criteria

- [ ] Read the ROS param server and hydrate to state
- [ ] Write to the ROS param server by sending a service request
- [ ] Develop a rudimentary UI
- [ ] Have UI dispatch actions to update the ROS param server and state

MathewMacDougall, 2019-04-27:
@kelvinkoon @qgolsteyn is this now an improved version of #392? Seems to cover similar stuff, but now with the new YAML

kelvinkoon, 2019-04-29:
> @kelvinkoon @qgolsteyn is this now an improved version of #392? Seems to cover similar stuff, but now with the new YAML

Yes, this is the updated version.

MathewMacDougall, 2019-04-29:
Ok, I'll close #392 then

PR #541 (2019-06-08): Communicate b/w ROS Param Server and Visualizer

## Please fill out the following before requesting review on this PR

### Description

This PR allows for reading and writing to the ROS param server using the Visualizer. The state is hydrated by reading from the /utils/ros/params .yaml files using js-yaml-loader. Writing to the ROS param server was done using the ROS util functions connected to the rudimentary UI.

### Testing Done

Testing was done manually by seeing how the ROS param server behaves after clicking start and stop on the Visualizer, as well as through the Redux Inspect tool in  [... cut]
closed 2019-06-08 by kelvinkoon