UBC-Thunderbots/Software #519: add buffer_time parameter to the findBestInterceptForBall function

MathewMacDougall, 2019-03-25:
### Description of the task

Add a `buffer_time` parameter to the `findBestInterceptForBall` function.

### Outline the task in more detail

The `buffer_time` parameter should specify the extra amount of time the robot should arrive at the ball intercept position before the ball gets there. Eg. a value of 1 second means the robot should arrive at the intercept location 1 second before the ball arrives. The default value of this parameter should be 0.

### Acceptance criteria

- [ ] Add the parameter to the function
- [ ] Unit tests
- [ ] Item C

MathewMacDougall, 2019-06-19:
@garethellis0 is this done?

garethellis0, 2019-06-19:
Looks like no, I'm not sure I was related to this? I'm also not sure this is needed, unless there is some specific thing requiring it that I don't know about.

MathewMacDougall, 2019-06-20:
Nope, just wanted to check since I wasn't sure and was cleaning up tickets

garethellis0, 2019-11-20:
@MathewMacDougall - can we close this?

MathewMacDougall, 2019-11-20:
Is it worth leaving around and linking to an Epic of all intercept improvements / ideas? I know we would likely revisit the functions since interception is a major goal this year, just don't want to lose this idea until we really get that rolling. Thoughts?

MathewMacDougall, 2019-11-20:
Ah I see for other related tickets these things should just come up naturally during discussion. In that case I'm fine closing this since I'm fairly confident we won't forget about this.

closed 2019-11-20 by MathewMacDougall