UBC-Thunderbots/Software #521: create "binary" utility function

MathewMacDougall, 2019-03-26:
### Description of the task

Create a new 'binary' function in `util/math_functions.*`

### Outline the task in more detail

The purpose of the function is to convert the input into "binary" data (0 or 1).

The function signature should be as follows:
`unsigned int binary(double value, double offset)`
`value` is the input to the function, and `offset` indicates where the step from 0 to 1 takes place (relative to 0). For example the result of `binary(0.1, 0)` would return `1`.

Basically, this is a unit step function.

### Acceptance criteria

- [ ] Implement the function
- [ ] unit test

FSXAC, 2019-04-10:
Would it make more sense if we called it `unitStep`?

MathewMacDougall, 2019-04-10:
Yea I think that's a good idea.

closed 2019-07-24 by MathewMacDougall