UBC-Thunderbots/Software #523: create "calcBinaryViolationScore" utility function for circles

MathewMacDougall, 2019-03-26:
### Description of the task

Create a violation function that indicates the violation score of a point with respect to a circle.

### Outline the task in more detail

Violation scores indicate how far into a "bad region" a given point is. The higher the score, the more the point is "violating" the region.

This function should return a value in the range [0,1] that indicates how much the given point is violating the region given by the circle. The function signature should be something like the following:
`unsigned int calcBinaryViolationScore(Circle c, Point p);`

The function should return a violation score of 1 if `p` is inside the circle, and 0 otherwise.

The `binary` function from #521 can be used for the implementation

### Acceptance criteria

- [ ] Implement the function
- [ ] unit test

FSXAC, 2019-04-10:
Why is blocked by the unit step function issue?

FSXAC, 2019-04-10:
Actually now that I come to think of it, once we implement `calcLinearTrespassScore`, couldn't the `calcBinaryTrespassScore` be like:

```
def calcBinaryTrespassScore(circle, point):
    return calcLinearTrespassScore(circle, point) > 0
```

MathewMacDougall, 2019-04-10:
It was blocked since we were originally planning to use the `unitStep` function as an implementation. You're right about the alternative implementation, but if we already have the `unitStep` function we might as well use that since it's a bit more consistent.

MathewMacDougall, 2019-07-24:
Closing because the current navigator doesn't need this. YAGNI

closed 2019-07-24 by MathewMacDougall