UBC-Thunderbots/Software #525: create "calcLinearViolationScore" utility function for circles

MathewMacDougall, 2019-03-26:
### Description of the task

Create a violation function that indicates the violation score of a point with respect to a circle.

### Outline the task in more detail

Violation scores indicate how far into a "bad region" a given point is. The higher the score, the more the point is "violating" the region.

This function should return a value in the range [0,1] that indicates how much the given point is violating the region given by the circle. The function signature should be something like the following:
`unsigned int calcLinearViolationScore(Circle c, Point p);`

The function should return a violation score of 0 if `p` is outside the circle, and values that linearly increase to 1 when the point is exactly in the middle of the circle.

The `linear` function from #522  can be used for the implementation

### Acceptance criteria

- [ ] Implement the function
- [ ] unit test

FSXAC, 2019-04-10:
the function prototype should be something like `double calcLinearTrespassScore(Circle c, Point p);` where the return value is a floating point right?

MathewMacDougall, 2019-04-10:
yup

MathewMacDougall, 2019-07-24:
Closing because the current navigator doesn't need this. YAGNI

closed 2019-07-24 by MathewMacDougall