UBC-Thunderbots/Software #580: Use autokick_fired to determine when the Kick Action is done

MathewMacDougall, 2019-04-29:
### Description of the task

The Kick Action currently checks the ball's velocity vector to determine if the robot has kicked the ball and the action is done. This should be upgraded to use the `autokick_fired` data from the robot status, since this will also very materialistically indicate if a kick has occured.

### Outline the task in more detail

The Kick Action should return done when the `autokick_fired` status has recently been true for the robot taking the kick.

### Acceptance criteria

- [ ] Use `autokick_fired` to calculate when the Kick Action is done
- [ ] Unit tests

garethellis0, 2019-11-20:
Marking as difficulty 13 because this will likely require some architecture changes to allow robot status to be exposed to actions.

jonathanlew, 2020-09-06:
Not worth the effort since checking vision for a kick (ball movement) should be sufficient

closed 2020-09-06 by jonathanlew